eerimoq/simba #9: Preemptive scheduler

eerimoq, 2016-05-18:
Implement a preemptive scheduler to meet hard real time requirements.

Currently a cooperative scheduler is implemented, and it works very well in most cases where the hard real time requirements can be implemented in interrupt context.

eerimoq, 2016-05-21:
A preemptive scheduler was implemtented for ARM in this commit: fd7cc905d10d2ca5634778cf5dd4109d0cc1e000

eerimoq, 2016-05-22:
Implemented on AVR. Will not implement on ESP.

eerimoq, 2016-05-23:
The ARM port does not work when rescheduling from the UART interrupt.

eerimoq, 2016-05-26:
Next attempt of the ARM preemptive scheduler looks promising. All tests pass. Closing this issue for now.

closed 2016-05-26 by eerimoq

rahuldeo2047, 2017-11-17:
Hi @eerimoq,
I want to understand ESP8266/32 can't have preemptive scheduler ports in **simba** framework. I believe esp rtos supports it.

> The ESP port won't make much sense since Simba is running as a FreeRTOS thread.
> Implemented on AVR. Will not implement on ESP.

:(

Is current cooperative scheduler enough for applications using sync, inet ( wifi, socket, tcp, data size 512B/32B ), i2c, gpio, pwm together with 5 ms deadline with precision and repeatability? Except for wifi and main-print, rest finishes within <<5 ms individually. However, still to profile. 5 ms can go up as driven by wifi-tx-rx-tcp. I can go till 30 - 40 ms with deterministic repeatability.

This is just an understanding request. And surely not criticizing why esp is not having the preemptive scheduler.

Currently, my application is not able to tx i2c collected data over tcp in targetted time. 
I am doing following things currently but not limited to:
1. main print / shell use -> thrd_respawn total 3
2. i2c thrd -> bus -> main and wifi
3. bus -> wifi thrd -> tcp tx -> tcp rx -> bus -> main / * and control * /
4. / * bus -> control thrd -> bus -> main and wifi * /

The jitter I am getting is about +/- 30 ms from x ms cycle. Sometimes it goes beyond 3 seconds as well. Maybe something to do with NO_DELAY kind of things in ESP end. Although, the host code can have performance issues.

The expectation from the application might be unrealistic. However, I want to push it.

Reference: 
https://github.com/PassionForRobotics/ATOM_DRONE/commit/bbcdcb80f6099f852b51e3b59cf97a2f9ab8a33e / ISSUE_#16/ros/rosatom and https://github.com/rahuldeo2047/ESP_rtos_eval/commit/e7a5c093db8d3b2acaf5093f5c688b915a9b77fb / ISSUE_#5/simba/i2c/sensors

If possible please let me know further steps, in case it is in context of **simba** repo.

Thank you

eerimoq, 2017-11-19:
I doubt that the ESP port will ever be deterministic. I would expect a lot of jitter, unfortunately. =/

rahuldeo2047, 2017-11-26:
What could be the reasons for jitters? Is it somehow due to the soc, Simba framework itself or something else. Hope to learn or get some direction towards this issue. :)

eerimoq, 2017-11-28:
Likely the WiFi stack will introduce some jitter for example. I have not tested this, so it's my best guess.