eerimoq/simba #14: Cygwin: "make -s console" fails with ImportError: No module named msvcrt

wingstar74, 2016-05-30:
$ make -s console
Traceback (most recent call last):
  File "../../simba/simba/make/console.py", line 6, in <module>
    import readchar
  File "/usr/lib/python2.7/site-packages/readchar/__init__.py", line 3, in <module>
    from .readchar import readchar, readkey
  File "/usr/lib/python2.7/site-packages/readchar/readchar.py", line 13, in <module>
    from .readchar_windows import readchar
  File "/usr/lib/python2.7/site-packages/readchar/readchar_windows.py", line 5, in <module>
    import msvcrt
ImportError: No module named msvcrt
../../simba/simba/make/app.mk:137: receptet för målet ”console” misslyckades
make: *** [console] Fel 1

Note that readchar is installed

eerimoq, 2016-05-30:
What about "ImportError: No module named msvcrt"? Maybe pip did not install that dependency?

eerimoq, 2016-05-30:
Meh, looks like readchar does not work in Cygwin. msvcrt is only available in Python for Windows.

eerimoq, 2016-05-30:
The problem is that readchar imports the windows version in Cygwin, when it should incluse the linux version instead. I manually modified /lib/python2.7/site-packages/readchar/readchar.py to import the linux version and then the console works.

eerimoq, 2016-05-30:
Copied readchar release 1.1.0 to the 3pp folder and made required changes. Committed in 2bf89cbfd1340f9c81a15fe7597c55a942f84680.

closed 2016-05-30 by eerimoq