eerimoq/simba #24: network_interface_wifi_station_espressif_start hang-up

wuwx, 2016-09-09:
if wifi ssid or password isn't correct
network_interface_wifi_station_espressif_start will blocked and device hang-up
is there some solution ?

eerimoq, 2016-09-09:
I made some changes to the network interface connection handling to give a solution to this. The start function no longer blocks until a connection has been established. The application can use the new ´´function network_interface_wifi_station_espressif_ip_up()´´ to check the interface status.

See this commit for the implementaiton: b4427156eaf8a3149b304d3d39ef6e22856686de

Note that that commit is not yet tested. I'll test it after work.

Let me know if you have other suggestions about how to improve the API.

Thanks.

wuwx, 2016-09-09:
i am using esp12e board,  platformio 3.0.1
main.c code like this, dhcp get ip address 192.168.201.117
but this code can not open http 8000 port
is there something i forgot to do ?
thank you :)

`#include "simba.h"

#include "esp_wifi.h"
#include "esp_sta.h"

static struct network_interface_wifi_station_espressif_t wifi;
static struct http_server_t server;
static struct shell_t shell;

THRD_STACK(listener_stack, 1024);
THRD_STACK(connection_stack, 1500);

static int request_index(struct http_server_connection_t *connection_p,
                         struct http_server_request_t *request_p);

static struct http_server_route_t routes[] = {
    { .path_p = "/", .callback = request_index },
    { .path_p = NULL, .callback = NULL }
};

static struct http_server_listener_t listener = {
    .address_p = "192.168.201.117",
    .port = 8000,
    .thrd = {
        .name_p = "http_listener",
        .stack = {
            .buf_p = listener_stack,
            .size = sizeof(listener_stack)
        }
    }
};

static struct http_server_connection_t connections[] = {
    {
        .thrd = {
            .name_p = "http_conn_0",
            .stack = {
                .buf_p = connection_stack,
                .size = sizeof(connection_stack)
            }
        }
    },

    { .thrd = { .name_p = NULL } }
};

static int request_index(struct http_server_connection_t *connection_p,
                         struct http_server_request_t *request_p)
{
    static const char index_html[] = "Simba";
    struct http_server_response_t response;

    /* Only the GET action is supported. */
    if (request_p->action != http_server_request_action_get_t) {
        return (-1);
    }

    /* Create the response. */
    response.code = http_server_response_code_200_ok_t;
    response.content.type = http_server_content_type_text_html_t;
    response.content.buf_p = index_html;
    response.content.size =
        strlen(response.content.buf_p);

    return (http_server_response_write(connection_p,
                                       request_p,
                                       &response));
}

static int no_route(struct http_server_connection_t *connection_p,
                    struct http_server_request_t *request_p)
{
    struct http_server_response_t response;

    /* Create the response. */
    response.code = http_server_response_code_404_not_found_t;
    response.content.type = http_server_content_type_text_html_t;
    response.content.buf_p = NULL;
    response.content.size = 0;

    return (http_server_response_write(connection_p,
                                       request_p,
                                       &response));
}

int main()
{
    sys_start();

    log_set_default_handler_output_channel(sys_get_stdout());

    inet_module_init();
    socket_module_init();
    network_interface_module_init();
    ping_module_init();

    network_interface_wifi_station_espressif_module_init();
    network_interface_wifi_station_espressif_init(&wifi, "ssid", "password");
    network_interface_add(&wifi.network_interface);
    network_interface_start(&wifi.network_interface);

    http_server_init(&server, &listener, connections, NULL, routes, no_route);
    http_server_start(&server);

    shell_init(&shell, sys_get_stdin(), sys_get_stdout(), NULL, NULL, "admin", "admin");
    shell_main(&shell);

    return (0);
}`

eerimoq, 2016-09-09:
The ip address of the TCP socket listening for connection is hard coded in the http server. I will fix that bug tonight.

Meanwhile, replace

https://github.com/eerimoq/simba/blob/master/src/inet/http_server.c#L325

with

``    inet_aton(listener_p->address_p, &addr.ip);``

so the ip address will be correct.

wuwx, 2016-09-09:
esp12e also support softap mode
is there some plan to add module support it ?

eerimoq, 2016-09-09:
No not for now. I'm all busy porting MicroPython to Simba ([Pumbaa](https://github.com/eerimoq/pumbaa)). Python on microcontrollers is quite nice =)

eerimoq, 2016-09-11:
A fix for the HTTP server problem has been implemented and will be included in future releases.

closed 2016-09-11 by eerimoq