eerimoq/simba #26: socket_sendto doesn't work for me

wuwx, 2016-09-09:
```
#include "simba.h"

static struct network_interface_wifi_station_espressif_t wifi;
static struct shell_t shell;

int main()
{
    sys_start();
        
    network_interface_module_init();
    socket_module_init();
    ping_module_init();

    network_interface_wifi_station_espressif_module_init();
    network_interface_wifi_station_espressif_init(&wifi, "SSID", "PASSWORD");
    network_interface_add(&wifi.network_interface);
    network_interface_start(&wifi.network_interface);
        
    shell_init(&shell, sys_get_stdin(), sys_get_stdout(), NULL, NULL, "admin", "admin");
    shell_main(&shell);

    uint8_t buf[16] = "hello";
    struct socket_t udp;
    struct inet_addr_t local_addr, remote_addr;
    socket_open_udp(&udp);
    inet_aton("192.168.38.149", &local_addr.ip);
    local_addr.port = 8888;
    inet_aton("192.168.38.249", &remote_addr.ip);
    remote_addr.port = 8888;
    socket_bind(&udp, &local_addr);
    for(int i = 0; i < 200; i++) {
        socket_sendto(&udp, buf, 16, 0,  &remote_addr);
        thrd_sleep_us(1000000);
    }
    socket_close(&udp);
        
    return (0);
}
```

wuwx, 2016-09-09:
```
$ inet/network_interface/list 
NAME            STATE  ADDRESS            TX BYTES    RX BYTES
esp-wlan        up     192.168.38.149            -           -
```

eerimoq, 2016-09-09:
``shell_main()`` start a shell in the current thread and never returns. Try to remove the two lines that initializes and starts the shell and the UDP socket should work.

The testcase for UDP looks like this, and it works on my setup:

https://github.com/eerimoq/simba/blob/master/tst/inet/network_interface/wifi_station_espressif/main.c#L83-L129

wuwx, 2016-09-09:
oh, if i craete a new thread to run udp send
i will be also ok ? is it right ?

eerimoq, 2016-09-09:
Yes that should work as well.

wuwx, 2016-09-09:
thank you very much~~~

eerimoq, 2016-09-09:
No problem. I'm happy someone actually find simba intresting =)

I'll do my best fixing issues and making improvements as the issues pop up. =)

closed 2016-09-11 by eerimoq