eerimoq/simba #27: compile error with cpp code

wuwx, 2016-09-10:
if i use arduino.cpp code like this:

```
#include <simba.h>
```

yes, it's a empty file, only include simba.h, then i use

`pio run`

i got many error message:

```
mbp:simba-heater wwx$ pio run
[Sat Sep 10 20:12:52 2016] Processing esp12e (platform: espressif8266, build_unflags: -Wno-implicit-function-declaration, build_flags: -DCONFIG_START_NETWORK=0, board: esp12e, framework: simba)
---------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 0 compatible libraries
Looking for dependencies...
Project does not have dependencies
Compiling .pioenvs/esp12e/src/arduino/arduino.o
cc1plus: warning: command line option '-std=gnu99' is valid for C/ObjC but not for C++ [enabled by default]
In file included from /Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/lwip/ip_addr.h:38:0,
from /Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/lwip/ip.h:39,
from /Users/wwx/.platformio/packages/framework-simba/src/inet/network_interface.h:26,
from /Users/wwx/.platformio/packages/framework-simba/src/simba.h:196,
from src/arduino/arduino.cpp:1:
/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/ipv4/lwip/ip4_addr.h:213:23: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]
LWIP_DEBUGF(debug, ("%"U16_F".%"U16_F".%"U16_F".%"U16_F,             \
^
/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/ipv4/lwip/ip4_addr.h:213:31: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]
LWIP_DEBUGF(debug, ("%"U16_F".%"U16_F".%"U16_F".%"U16_F,             \
^
/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/ipv4/lwip/ip4_addr.h:213:40: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]
LWIP_DEBUGF(debug, ("%"U16_F".%"U16_F".%"U16_F".%"U16_F,             \
^
/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/ipv4/lwip/ip4_addr.h:213:49: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]
LWIP_DEBUGF(debug, ("%"U16_F".%"U16_F".%"U16_F".%"U16_F,             \
^
In file included from /Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/lwip/ip_addr.h:39:0,
from /Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/lwip/ip.h:39,
from /Users/wwx/.platformio/packages/framework-simba/src/inet/network_interface.h:26,
from /Users/wwx/.platformio/packages/framework-simba/src/simba.h:196,
from src/arduino/arduino.cpp:1:
/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/ipv6/lwip/ip6_addr.h:263:23: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]
LWIP_DEBUGF(debug, ("%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F, \
^
/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/ipv6/lwip/ip6_addr.h:263:31: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]
LWIP_DEBUGF(debug, ("%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F, \
^
/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/ipv6/lwip/ip6_addr.h:263:40: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]
LWIP_DEBUGF(debug, ("%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F, \
^
/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/ipv6/lwip/ip6_addr.h:263:49: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]
LWIP_DEBUGF(debug, ("%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F, \
^
/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/ipv6/lwip/ip6_addr.h:263:58: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]
LWIP_DEBUGF(debug, ("%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F, \
^
/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/ipv6/lwip/ip6_addr.h:263:67: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]
LWIP_DEBUGF(debug, ("%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F, \
^
/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/ipv6/lwip/ip6_addr.h:263:76: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]
LWIP_DEBUGF(debug, ("%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F, \
^
/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/ipv6/lwip/ip6_addr.h:263:85: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]
LWIP_DEBUGF(debug, ("%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F":%"X16_F, \
^
In file included from /Users/wwx/.platformio/packages/framework-simba/src/simba.h:44:0,
from src/arduino/arduino.cpp:1:
[... 143 lines cut ...]
/Users/wwx/.platformio/packages/framework-simba/src/sync/chan.h:28:14: note: 'chan_t' has a previous declaration here
typedef void chan_t;
^
In file included from /Users/wwx/.platformio/packages/framework-simba/src/simba.h:191:0,
from src/arduino/arduino.cpp:1:
/Users/wwx/.platformio/packages/framework-simba/src/inet/socket.h:26:12: error: using typedef-name 'chan_t' after 'struct'
struct chan_t base;
^
In file included from /Users/wwx/.platformio/packages/framework-simba/src/simba.h:56:0,
from src/arduino/arduino.cpp:1:
/Users/wwx/.platformio/packages/framework-simba/src/sync/chan.h:28:14: note: 'chan_t' has a previous declaration here
typedef void chan_t;
^
In file included from /Users/wwx/.platformio/packages/framework-simba/src/inet/network_interface/wifi_station_espressif.h:26:0,
from /Users/wwx/.platformio/packages/framework-simba/src/simba.h:201,
from src/arduino/arduino.cpp:1:
/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/espressif/esp_sta.h:360:6: error: use of enum 'dhcp_status' without previous declaration
enum dhcp_status wifi_station_dhcpc_status(void);
^
/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/espressif/esp_sta.h:360:49: error: invalid type in declaration before ';' token
enum dhcp_status wifi_station_dhcpc_status(void);
^
*** [.pioenvs/esp12e/src/arduino/arduino.o] Error 1
====================================================== [ERROR] Took 2.26 seconds ======================================================
```

eerimoq, 2016-09-10:
That's expected. Only c is supported. As many libraries on the internet are written in c++ I'll probably be forced to support that language, even if I personally like c better.

wuwx, 2016-09-10:
because i hope can use simba with some arduino library write in c++
so .......

eerimoq, 2016-09-10:
Okay first I have to understand the impact of supporting c++. It should be fairly easy to support. A few extern c and maybe change a few type names for the c++ compiler to accept the c headers. What happens if you add

```
extern "C" {
    #include "simba.h"
 }
```

 in the cpp file? I cannot test it because I'm not at my computer.

eerimoq, 2016-09-15:
Support for c++ compilation has been implemented. It's somewhat messy, but should work for now.

closed 2016-09-15 by eerimoq