eerimoq/simba #30: spiffs test unpassed on esp12e board

wuwx, 2016-09-15:
the result like this:
it's broken on test_read_write 

```
================================== TEST BEGIN ==================================

app:    app_name-8.0.0 built 2016-09-15 17:24 CST by wwx.
board:  ESP-12E Development Board
mcu:    Espressif ESP8266 @ 80 MHz, 82 kB dram, 4 MB flash

enter: test_init
exit: test_init: PASSED

enter: test_format
exit: test_format: PASSED

enter: test_read_write
�@*r��P���K�t�dr�-@��.�Ѫ�½�a �@*r��P���K�t�dr�.@��.�Ѫ�½�a /t�$v'

```

use this test code:
https://github.com/eerimoq/simba/tree/master/tst/filesystems/spiffs

eerimoq, 2016-09-15:
I'll have a look at it. It's probably a memory alignment issue. There are some warnings in spiffs when compiling the code related to this.

Just to be clear, the spiffs in the test case does only use a real flash memory for Arduino Due. For other boards the storage is in RAM. I guess you want to use the spiffs on a ESP with storage in flash, and as the flash driver is not yet supported on ESP no persistent storage is available. I'll have a look at it right away and implement it if is it not too complicated. Should be fairly easy.

wuwx, 2016-09-15:
yes, i want storage json config to flash stoage~~~~~~
i love simba !!!!!!

eerimoq, 2016-09-15:
thanks =) I'm quite fond of it myself!

The ESP port is probably the most important port as it is a great chipset taking over the market, but unfortunately I've been implementing a lot of functions in that port. I've mostly worked on AVR and ARM. It's great that you help out improving the ESP port! =)

eerimoq, 2016-09-16:
Fixed in d0c2429a352d6c6c89b8f818e640fe20d6027a0f. SPIFFS data was not aligned and memory access violations occured.

closed 2016-09-16 by eerimoq