eerimoq/simba #33: SPIFFS data lost on esp12e after reset or reboot

wzk, 2016-09-16:
Thank you for this great project of simba !

I use test code (locate on 'tst/filesystems/spiffs/' ) to verify this issue. 

I change 

>  BTASSERT(spiffs_format(&fs) == 0);

to 

>    if ( spiffs_mount(&fs, &config, workspace, fdworkspace, sizeof(fdworkspace),  cache,  sizeof(cache), NULL) == SPIFFS_ERR_NOT_A_FS) {
>         std_printf("Formatting file system\n");
>         BTASSERT(spiffs_format(&fs) == 0);
>    }

in `test_format(struct harness_t *harness_p)' function, and I get this result after reset the ESP.

> enter: test_format
> Formatting file system
> exit: test_format: PASSED

Does SPIFFS need to format file system every time on boot under ESP12E?

eerimoq, 2016-09-16:
Glad you like simba :)

The problem is that the test suite does not store data in flash, but in RAM. Yesterday I started to implement the flash driver needed to store data persistently in the flash of the ESP. If the implementation progresses as planned it will be ready during the weekend.

wzk, 2016-09-17:
Thank you for your hard work!

eerimoq, 2016-09-18:
A first draft of a flash driver for ESP is implemented and seems to work well.

When uploading software to the ESP using the esptool the file system will not be erased. use ``fs_format()`` or the file system command ``filesystem/fs/format <path>`` to format it.

closed 2016-09-18 by eerimoq