eerimoq/simba #34: webserver with large http data is not work

wuwx, 2016-09-22:
is esp8266, I setup a http server, listener give stack 2048, http_connection give stack 8192.
if http response a large http data, like 5KB html page.
then the client can't open the page normal.

is that because socket_write size bigger then TCP_SND_BUF ?

eerimoq, 2016-09-22:
I'm not sure what the problem is. It is possible to only write the HTTP header with [http_server_response_write()](https://github.com/eerimoq/simba/blob/master/src/inet/http_server.h#L178-L195) and then write the page contents with [socket_write()](https://github.com/eerimoq/simba/blob/master/src/inet/socket.h#L196-L209).

See the test code here: https://github.com/eerimoq/simba/blob/master/tst/inet/http_server/main.c#L92-L123

wuwx, 2016-09-23:
i can't run the test code ......

```╭─wwx@carbon /tmp/hello  
╰─➤  pio run
[Fri Sep 23 10:34:27 2016] Processing esp12e (platform: espressif8266, board: esp12e, framework: simba)
----------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 0 compatible libraries
Looking for dependencies...
Project does not have dependencies
Linking .pioenvs/esp12e/firmware.elf
.pioenvs/esp12e/src/main.o:(.text.test_request_websocket+0x20): undefined reference to `socket_stub_accept'
.pioenvs/esp12e/src/main.o:(.text.test_request_websocket+0x24): undefined reference to `socket_stub_input'
.pioenvs/esp12e/src/main.o:(.text.test_request_websocket+0x28): undefined reference to `socket_stub_output'
.pioenvs/esp12e/src/main.o:(.text.test_request_websocket+0x41): undefined reference to `socket_stub_accept'
.pioenvs/esp12e/src/main.o: In function `test_request_websocket':
main.c:(.text.test_request_websocket+0x4d): undefined reference to `socket_stub_input'
main.c:(.text.test_request_websocket+0x58): undefined reference to `socket_stub_output'
main.c:(.text.test_request_websocket+0xaf): undefined reference to `socket_stub_input'
main.c:(.text.test_request_websocket+0xb9): undefined reference to `socket_stub_output'
.pioenvs/esp12e/src/main.o:(.text.test_request_no_route+0x14): undefined reference to `socket_stub_accept'
.pioenvs/esp12e/src/main.o:(.text.test_request_no_route+0x1f): undefined reference to `socket_stub_input'
.pioenvs/esp12e/src/main.o:(.text.test_request_no_route+0x2a): undefined reference to `socket_stub_output'
.pioenvs/esp12e/src/main.o:(.text.test_request_index+0x14): undefined reference to `socket_stub_accept'
.pioenvs/esp12e/src/main.o:(.text.test_request_index+0x1f): undefined reference to `socket_stub_input'
.pioenvs/esp12e/src/main.o:(.text.test_request_index+0x2a): undefined reference to `socket_stub_output'
.pioenvs/esp12e/src/main.o:(.text.test_start+0x18): undefined reference to `socket_stub_init'
.pioenvs/esp12e/src/main.o:(.text.test_start+0x2a): undefined reference to `socket_stub_init'
collect2: error: ld returned 1 exit status
*** [.pioenvs/esp12e/firmware.elf] Error 1
================================================ [ERROR] Took 1.99 seconds ================================================

╭─wwx@carbon /tmp/hello  
╰─➤  pio run
[Fri Sep 23 10:34:27 2016] Processing esp12e (platform: espressif8266, board: esp12e, framework: simba)
----------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 0 compatible libraries
Looking for dependencies...
Project does not have dependencies
Linking .pioenvs/esp12e/firmware.elf
.pioenvs/esp12e/src/main.o:(.text.test_request_websocket+0x20): undefined reference to `socket_stub_accept'
.pioenvs/esp12e/src/main.o:(.text.test_request_websocket+0x24): undefined reference to `socket_stub_input'
.pioenvs/esp12e/src/main.o:(.text.test_request_websocket+0x28): undefined reference to `socket_stub_output'
.pioenvs/esp12e/src/main.o:(.text.test_request_websocket+0x41): undefined reference to `socket_stub_accept'
.pioenvs/esp12e/src/main.o: In function `test_request_websocket':
main.c:(.text.test_request_websocket+0x4d): undefined reference to `socket_stub_input'
main.c:(.text.test_request_websocket+0x58): undefined reference to `socket_stub_output'
main.c:(.text.test_request_websocket+0xaf): undefined reference to `socket_stub_input'
main.c:(.text.test_request_websocket+0xb9): undefined reference to `socket_stub_output'
.pioenvs/esp12e/src/main.o:(.text.test_request_no_route+0x14): undefined reference to `socket_stub_accept'
.pioenvs/esp12e/src/main.o:(.text.test_request_no_route+0x1f): undefined reference to `socket_stub_input'
.pioenvs/esp12e/src/main.o:(.text.test_request_no_route+0x2a): undefined reference to `socket_stub_output'
.pioenvs/esp12e/src/main.o:(.text.test_request_index+0x14): undefined reference to `socket_stub_accept'
.pioenvs/esp12e/src/main.o:(.text.test_request_index+0x1f): undefined reference to `socket_stub_input'
.pioenvs/esp12e/src/main.o:(.text.test_request_index+0x2a): undefined reference to `socket_stub_output'
.pioenvs/esp12e/src/main.o:(.text.test_start+0x18): undefined reference to `socket_stub_init'
.pioenvs/esp12e/src/main.o:(.text.test_start+0x2a): undefined reference to `socket_stub_init'
collect2: error: ld returned 1 exit status
*** [.pioenvs/esp12e/firmware.elf] Error 1
================================================ [ERROR] Took 1.99 seconds ================================================
```

eerimoq, 2016-09-23:
The file ``socket_stub.c`` must be compiled...

wuwx, 2016-09-27:
i try this code from example, if only one line: "Hello from Simba!", it works fine.
but if we add more "Hello from Simba" string, then use curl get info:

```
$ curl http://192.168.1.9:8000/index.html -v
*   Trying 192.168.1.9...
* Connected to 192.168.1.9 (192.168.1.9) port 8000 (#0)
> GET /index.html HTTP/1.1
> Host: 192.168.1.9:8000
> User-Agent: curl/7.43.0
> Accept: */*
> 
< HTTP/1.1 200 OK
< Content-Type: text/html
< Content-Length: 4866
< 
* transfer closed with 4866 bytes remaining to read
* Closing connection 0
curl: (18) transfer closed with 4866 bytes remaining to read

```

```
/**
 * @file main.c
 *
 * @section License
 * Copyright (C) 2016, Erik Moqvist
 *
 * This library is free software; you can redistribute it and/or
 * modify it under the terms of the GNU Lesser General Public
 * License as published by the Free Software Foundation; either
 * version 2.1 of the License, or (at your option) any later version.
 *
 * This library is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 * Lesser General Public License for more details.
 *
 * This file is part of the Simba project.
 */

#include "simba.h"

static struct http_server_t server;
static struct shell_t shell;

THRD_STACK(listener_stack, 1024);
THRD_STACK(connection_stack, 1500);

static int request_index(struct http_server_connection_t *connection_p,
                         struct http_server_request_t *request_p);

static struct http_server_route_t routes[] = {
    { .path_p = "/index.html", .callback = request_index },
    { .path_p = NULL, .callback = NULL }
};

static struct http_server_listener_t listener = {
    .address_p = "0.0.0.0",
    .port = 8000,
    .thrd = {
        .name_p = "http_listener",
        .stack = {
            .buf_p = listener_stack,
            .size = sizeof(listener_stack)
        }
    }
};

[... 228 lines cut ...]

    http_server_start(&server);

    return (0);
}

#endif

int main()
{
    init();

    shell_init(&shell,
               sys_get_stdin(),
               sys_get_stdout(),
               NULL,
               NULL,
               NULL,
               NULL);
    shell_main(&shell);

    return (0);
}

```

eerimoq, 2016-09-27:
Does it work if you send less than 4096 bytes?

wuwx, 2016-09-27:
```
$ curl http://192.168.1.9:8000/index.html -v                                      
*   Trying 192.168.1.9...
* Connected to 192.168.1.9 (192.168.1.9) port 8000 (#0)
> GET /index.html HTTP/1.1
> Host: 192.168.1.9:8000
> User-Agent: curl/7.43.0
> Accept: */*
> 
< HTTP/1.1 200 OK
< Content-Type: text/html
< Content-Length: 3970
< 
* transfer closed with 3970 bytes remaining to read
* Closing connection 0
curl: (18) transfer closed with 3970 bytes remaining to read
```

wuwx, 2016-09-28:
send less than 4096 bytes (3970) is also not work

eerimoq, 2016-09-28:
What's the maximum number of byte that works?

wuwx, 2016-09-28:
this is the code can work with max html body

```
/**
 * @file main.c
 *
 * @section License
 * Copyright (C) 2016, Erik Moqvist
 *
 * This library is free software; you can redistribute it and/or
 * modify it under the terms of the GNU Lesser General Public
 * License as published by the Free Software Foundation; either
 * version 2.1 of the License, or (at your option) any later version.
 *
 * This library is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 * Lesser General Public License for more details.
 *
 * This file is part of the Simba project.
 */

#include "simba.h"

static struct http_server_t server;
static struct shell_t shell;

THRD_STACK(listener_stack, 1024);
THRD_STACK(connection_stack, 15000);

static int request_index(struct http_server_connection_t *connection_p,
                         struct http_server_request_t *request_p);

static struct http_server_route_t routes[] = {
    { .path_p = "/index.html", .callback = request_index },
    { .path_p = NULL, .callback = NULL }
};

static struct http_server_listener_t listener = {
    .address_p = "0.0.0.0",
    .port = 8000,
    .thrd = {
        .name_p = "http_listener",
        .stack = {
            .buf_p = listener_stack,
            .size = sizeof(listener_stack)
        }
    }
};

static struct http_server_connection_t connections[] = {
    {
        .thrd = {
            .name_p = "http_conn_0",
            .stack = {
                .buf_p = connection_stack,
                .size = sizeof(connection_stack)
            }
        }
    },

    { .thrd = { .name_p = NULL } }
};

/**
 * Handler for the index request.
 */
static int request_index(struct http_server_connection_t *connection_p,
                         struct http_server_request_t *request_p)
{
[... 191 lines cut ...]
               NULL);
    shell_main(&shell);

    return (0);
}

```

```
$ curl http://192.168.1.9:8000/index.html -v
*   Trying 192.168.1.9...
* Connected to 192.168.1.9 (192.168.1.9) port 8000 (#0)
> GET /index.html HTTP/1.1
> Host: 192.168.1.9:8000
> User-Agent: curl/7.43.0
> Accept: */*
> 
< HTTP/1.1 200 OK
< Content-Type: text/html
< Content-Length: 2854
< 
<!DOCTYPE HTML>
......
......
```

wuwx, 2016-09-28:
the http body size = 2854
the http header size = 66 （each line has \r\n）
the TCP_MSS = 1460
1460 * 2 == 2854 + 66
ha ha ha ha

eerimoq, 2016-10-08:
Implemented on master. Socket module rewritten. =)

closed 2016-10-08 by eerimoq