eerimoq/simba #35: if boot ESP12e board, you can see message: formating

wuwx, 2016-09-22:
if set CONFIG_START_FILESYSTEM = 1
then formating the fs every boot time

eerimoq, 2016-09-22:
Hmm... that sounds bad. The file system logic first tries to mount the file system, and only if it fails it formats it.

https://github.com/eerimoq/simba/blob/master/src/kernel/sys/filesystem.i#L205-L240

Can you provide logs of multiple boots?

eerimoq, 2016-09-22:
Please run ``/kernel/sys/config`` and ``/kernel/sys/info`` in the shell. It will print information about the file system configuration.

wuwx, 2016-09-22:
```
config: sys-config-string=1
        assert=1
        debug=1
        fs-cmd-ds18b20-list=1
        fs-cmd-fs-counters-list=1
        fs-cmd-fs-counters-reset=1
        fs-cmd-fs-filesystems-append=1
        fs-cmd-fs-filesystems-list=1
        fs-cmd-fs-filesystems-read=1
        fs-cmd-fs-filesystems-write=1
        fs-cmd-fs-parameters-list=1
        fs-cmd-i2c-read=1
        fs-cmd-i2c-write=1
        fs-cmd-log-list=1
        fs-cmd-log-print=1
        fs-cmd-log-set-log-mask=1
        fs-cmd-network-interface-list=1
        fs-cmd-pin-read=1
        fs-cmd-pin-set-mode=1
        fs-cmd-pin-write=1
        fs-cmd-settings-list=1
        fs-cmd-settings-read=1
        fs-cmd-settings-reset=1
        fs-cmd-settings-write=1
        fs-cmd-sys-info=1
        fs-cmd-sys-uptime=1
        fs-cmd-thrd-list=1
        fs-cmd-thrd-set-log-mask=1
        fs-cmd-usb-device-list=1
        fs-cmd-usb-host-list=1
        monitor-thread=0
        preemptive-scheduler=0
        profile-stack=1
        settings-area-size=256
        shell-command-max=64
        shell-history-size=768
        shell-minimal=0
        shell-prompt="$ "
        spiffs=1
        start-console=1
        start-console-device-index=0
        start-console-uart-baudrate=74880
        start-console-usb-cdc-control-interface=0
        start-console-usb-cdc-endpoint-in=2
        start-console-usb-cdc-endpoint-out=3
        start-console-usb-cdc-wait-for-connetion=1
        start-filesystem=1
        start-filesystem-address=0x00300000
        start-filesystem-size=0x100000
        start-network=0
        start-network-interface-wifi-ssid=SSID
        start-network-interface-wifi-password=********
        start-shell=1
        start-shell-prio=30
        start-shell-stack-size=1536
        std-output-buffer-max=16
        system-tick-frequency=100
        usb-device-vid=0x2341
        usb-device-pid=0x8037

```

```
/kernel/sys/info
app:    app_name-8.0.0 built 2016-09-22 23:59 CST by wwx.
board:  ESP-12E Development Board
mcu:    Espressif ESP8266 @ 80 MHz, 82 kB dram, 4 MB flash
```

wuwx, 2016-09-22:
may be i know why
the last 20k space used to eeprom
so ......
i can post a pull request

eerimoq, 2016-09-22:
The fix in the pull request fixes this issue, right?

wuwx, 2016-09-23:
yes, it's woking now~~~~

closed 2016-09-23 by eerimoq