eerimoq/simba #37: Http Server do not support POST method

wzk, 2016-09-22:
I find those lines in http_server.c file:

    /* Save the action and path in the request struct. */
    strcpy(request_p->path, path_p);

    if (strcmp(action_p, "GET") == 0) {
        request_p->action = http_server_request_action_get_t;
    } else {
        return (-1);
    }

but, in most cases, POST method is used to submit form in most case,  I hope POST method will be impliment in simba project.

eerimoq, 2016-09-22:
I'll see what I can do. I'm currently porting MicroPython on Simba ([Pumbaa](https://github.com/eerimoq/pumbaa)) to the ESP8266 and it has taken longer than expected. I should be fairly close to have it finalized.

Feel free to implement the POST method and be part of the Simba Project =)

eerimoq, 2016-09-23:
No there is no such function.

I implemented the POST action in the HTTP server. Have a look at c6c680220cf4ffe890069d2dd038c7e585d6c6a7.

closed 2016-09-23 by eerimoq