eerimoq/simba #46: ESP8266 AP Mode

benpeoples, 2016-10-09:
Would love to see support for AP mode in ESP8266.

We use AP mode as a captive configuration portal a bunch: i.e., if it can't find the right SSID to connect to, it'll just create an AP.   Best practice is of course to periodically scan for the SSID again.

Anyway, I may be volunteering to develop this, but I'm swamped for a few weeks.

eerimoq, 2016-10-09:
I would gladly accept any help improving the networking modules in Simba (any any other functionality, of course).

As networking is already part of many embedded applications and the demand will increase in the future (ESP32 and other modules) it's important to give the user a network stack that is easy to use. I think I'm on the right track with network interfaces and I've planned to implement a network manager that monitors the interfaces and tries to reconnect to WiFi:s if the connection is lost, just as any normal operating system does.

It's quite a challenge to get this right =)

benpeoples, 2016-10-09:
And there's already a pull request, neat.   

This kind of thing is really challenging, but I like your philosophy on it.  =)

eerimoq, 2016-10-15:
There is a first implementation if WiFi AP for the ESP committed and pushed.

See:

https://github.com/eerimoq/simba/blob/master/src/drivers/espressif_wifi.h
https://github.com/eerimoq/simba/tree/master/src/drivers/espressif_wifi

closed 2016-10-15 by eerimoq