eerimoq/simba #51: HTTP server in one thread

eerimoq, 2016-10-11:
The HTTP server currently has one listener thread and on or more client threads. This is memory consuming and it should be considered to only have one thread that both listens for connections and acts on requests. It's unlikely that more than one client will be active at a time in small systems.

wuwx, 2016-10-12:
if one http server html page
and one web_socket server
it's seems need 2 connections ?

eerimoq, 2016-10-12:
The function ``chan_list_poll()`` can be used to poll multiple sockets for input data. The idea is to use that function in the top loop of a HTTP Server thread to poll the listener socket, "client" sockets and all websockets.

wuwx, 2016-10-12:
if use one thread work well
may free 1K memery
it's a good news

closed 2017-02-18 by eerimoq