eerimoq/simba #52: how can i send and recv udp data use one socket

wuwx, 2016-10-16:
I try to use the UDP protocol for the communication between the client and the server.
The client sends a message to the server periodically, and the server may also send messages to the UDP port, which is monitored by the client.

The client is behand in a NAT server.

So I used two thrd to carry out this operation, a thrd dedicated users to monitor the UDP port data, another thrd dedicated cycle of sending messages.

At the same time, in order to allow the client to send the port and the server to send the port is the same socket, I put this socket is set to the global variable

At the beginning of a simple client to send a message when quite normal, but once the server to respond to a message, immediately went wrong.

I guess it is because socket_send_to and socket_recvfrom will change the thrd_p element of the socket structure.

Do not know how to achieve this kind of demand is better?

```
//global
struct socket_t udp;

// thrd_1
socket_open_udp(&udp);
socket_bind(&udp, &local_addr);
while(1) {
        socket_recvfrom(&udp, buf, 31, 0, &remote_server_addr);
}

// thrd_2

while(1) {
        socket_sendto(&udp, buf, strlen(buf), 0,  &remote_server_addr);
        thrd_sleep(10);
}

```

eerimoq, 2016-10-16:
Yes you're right, two threads may not use the same socket simultaneously. The application will crash.

Here are two alternative implementations:

1. Create two sockets, one for receiving data and one for transmitting data. Bind the receiving socket to the local address, and do *not* bind the transmitting socket. I have not tested it but I think it will work. Depends on how LwIP handles this use case.

2. Create two sockets and bind them to different local ports, if that works with your use case.

So the key is to create multiple sockets and somehow make that fit into the application. I think it's hard to allow multiple threads to use the same socket. Maybe it is, but I have to think about it. =)

wuwx, 2016-10-16:
The client is at the back of the NAT server, which means that I have to use the same listening port, to allow the server to respond to the data back to the client to handle correctly
If the client uses a different socket or is not bound to a fixed address
Can cause the server to respond directly to the data can not be received, so I also feel very difficult

eerimoq, 2016-10-16:
Hmm, tricky... How is this normally implemented?

Would it work to have one thread and check if there is input data interleaved with sending packets? ``socket_size()`` should return a non-zero value if there is a UDP packet available in the input queue. Unfortunately there is no possibility to configure a socket as non-blocking as of today.

eerimoq, 2016-10-16:
Is it allowed to call LwIP functions, like allocating a pbuf or sending a udp packet, from all threads in an  application? I'm not comfortable doing that from the Simba thread as I do not know how LwIP works. Maybe it is ok?

wuwx, 2016-10-16:
I tested it, it seems to work, but this practice is not beautiful, It's really not so comfortable
, but it can only be so, or I have to switch to a espconn_sendto function

thanks ~~~~~

eerimoq, 2016-10-16:
Another alternative is to put ``thrd_p`` on the stack of the sending/receiving thread. That way it will be unique for each thread.

eerimoq, 2016-10-16:
Okay, I just pushed a fix! Let me know if there is still a problem!

Note: Only one sender and one reader may be using the socket at the same time. Two or more senders and/or receivers are not allowed.

wuwx, 2016-10-17:
yes, it's working now
nice ~~~, thanks :100:

closed 2016-10-17 by eerimoq