eerimoq/simba #53: ESP32 port

eerimoq, 2016-10-17:
The ESP32 is the successor to the ESP8266. It is a dual core MCU with lots of RAM and ROM. The SDK runs on FrooRTOS and there are an Arduino core and IoT framework already being developed by Espressif and the ESP community.

Take the same approach as for ESP8266; run Simba in an FreeRTOS thread. It's the only implementation that can be accomplished in a reasonable amount of time.

Also consider not porting at all, because the Espressif SDK and Arduino are very good alternatives for most users.

Steps in porting Simba to the ESP32.

1. Copy Espressif SDK header files and libraries to 3pp/.
2. Add ESP32 to mcus/ and boards/.
3. Implement ``main()`` and compile, upload and run. Build system changes will be required.
4. Run Simba in a FreeRTOS thread.
5. Adapt the WiFi and socket modules. Is hopefully similar to ESP8266.
6. ...

wuwx, 2016-10-17:
wow~~~, good luck for  you ~~~

wuwx, 2016-10-17:
@eerimoq what esp32 board are you use ?

eerimoq, 2016-10-17:
I don't have a ESP32 board yet. Can you recommend one? Any good chinese sites selling them that delivers world wide?

wuwx, 2016-10-17:
I also do not have the esp32 board
the esp32 soc is release
but the dev board not released

eerimoq, 2016-10-17:
I might start the port anyway just to make it compile and link. Then it's just a matter of uploading it to the board when I get one. If I'm lucky it works immediately. :P

eerimoq, 2016-10-17:
Port started on branch [esp32](https://github.com/eerimoq/simba/tree/esp32)!

[esp-idf](https://github.com/espressif/esp-idf) has been added as a git submodule in ``3pp/`` and the mcu and boards folders have been created (based on the ESP8266 port).

It actually compiles and links, but there are many files in the esp-idf that are not yet added to the list of source files. So the binary will by no means run on the microcontroller.

eerimoq, 2016-10-22:
The esp32 branch has been merged to master. Still waiting for a NANO32 board to test on.

eerimoq, 2016-10-29:
The initial port has been implemented and pushed. The kernel package, UART driver and pin driver has been ported.

closed 2016-10-29 by eerimoq