eerimoq/simba #54: platform io esp8266 broken

wuwx, 2016-10-23:
in commit: 5d724e373d852603b81a1e7a5be7b273d70f1e78
esp8266 is not work~~~~~
main.c code

```
#include "simba.h"
int main()
{
    sys_start();
    return 0;
}
```

wuwx, 2016-10-23:
maybe because lose
```
-Wl,--start-group -lhal -lphy -lpp -lnet80211 -lwpa -lcrypto -lmain -lfreertos -llwip -Wl,--end-group
```
before
```
-Wl,-Tsimba.flash.4m.ld
```

eerimoq, 2016-10-23:
Are you building with platformio or the simba build system? Can you show the verbose build log?

The sys test suite build and uploaded with platformio passes on my setup on an ESP-12E.

wuwx, 2016-10-23:
i use platformio

```
pio run -v                                                                                                                                     1 ↵
[Sun Oct 23 15:51:49 2016] Processing esp12e (platform: espressif8266, board: esp12e, framework: simba)
--------------------------------------------------------------------------------------------------------------------------------------------------------
Collected 0 compatible libraries
Looking for dependencies...
Project does not have dependencies
xtensa-lx106-elf-gcc -o .pioenvs/esp12e/src/main.o -c -fdata-sections -fno-inline-functions -Werror -c -Os -std=gnu99 -funsigned-bitfields -mlongcalls -mtext-section-literals -Wall -nostdlib -fsingle-precision-constant -Wno-error=unused-variable -ffunction-sections -Os -mlongcalls -mtext-section-literals -falign-functions=4 -ffunction-sections -fdata-sections -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=030100 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DARCH_ESP -DFAMILY_ESP -DMCU_ESP8266 -DBOARD_ESP12E -DVERSION=esp32 -DICACHE_FLASH -D__STRICT_ANSI__ -D__ets__ -U__STRICT_ANSI__ -I.pioenvs/esp12e/SimbaFramework/src -I/Users/wwx/.platformio/packages/framework-simba/src -I.pioenvs/esp12e/SimbaFramework/3pp/ESP8266_RTOS_SDK/include -I/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include -I.pioenvs/esp12e/SimbaFramework/build/esp12e -I/Users/wwx/.platformio/packages/framework-simba/build/esp12e -I.pioenvs/esp12e/SimbaFramework/3pp/ESP8266_RTOS_SDK/include/espressif/esp8266 -I/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/espressif/esp8266 -I/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/espressif/esp8266 -I/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/espressif/esp8266 -I.pioenvs/esp12e/SimbaFramework/3pp/spiffs-0.3.5/src -I/Users/wwx/.platformio/packages/framework-simba/3pp/spiffs-0.3.5/src -I.pioenvs/esp12e/SimbaFramework/3pp/ESP8266_RTOS_SDK/extra_include -I/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/extra_include -I.pioenvs/esp12e/SimbaFramework/src/drivers/ports/esp -I/Users/wwx/.platformio/packages/framework-simba/src/drivers/ports/esp -I.pioenvs/esp12e/SimbaFramework/3pp/ESP8266_RTOS_SDK/include/espressif -I/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/espressif -I/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/espressif -I.pioenvs/esp12e/SimbaFramework/src/boards/esp12e -I/Users/wwx/.platformio/packages/framework-simba/src/boards/esp12e -I/Users/wwx/.platformio/packages/framework-simba/src/boards/esp12e -I.pioenvs/esp12e/SimbaFramework/src/kernel/ports/esp/gnu -I/Users/wwx/.platformio/packages/framework-simba/src/kernel/ports/esp/gnu -I.pioenvs/esp12e/SimbaFramework/3pp/ESP8266_RTOS_SDK/include/lwip -I/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip -I/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip -I.pioenvs/esp12e/SimbaFramework/3pp/ESP8266_RTOS_SDK/include/lwip/ipv6 -I/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/ipv6 -I/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/ipv6 -I/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/ipv6 -I.pioenvs/esp12e/SimbaFramework/3pp/ESP8266_RTOS_SDK/include/lwip/ipv4 -I/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/ipv4 -I/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/ipv4 -I/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/lwip/ipv4 -I.pioenvs/esp12e/SimbaFramework/src/inet -I/Users/wwx/.platformio/packages/framework-simba/src/inet -I/Users/wwx/.platformio/packages/framework-simba/src/inet -I.pioenvs/esp12e/SimbaFramework/src/mcus/esp8266 -I/Users/wwx/.platformio/packages/framework-simba/src/mcus/esp8266 -I/Users/wwx/.platformio/packages/framework-simba/src/mcus/esp8266 -Isrc src/main.c
xtensa-lx106-elf-gcc -o .pioenvs/esp12e/SimbaFramework/3pp/spiffs-0.3.5/src/spiffs_cache.o -c -fdata-sections -fno-inline-functions -Werror -c -Os -std=gnu99 -funsigned-bitfields -mlongcalls -mtext-section-literals -Wall -nostdlib -fsingle-precision-constant -Wno-error=unused-variable -ffunction-sections -Os -mlongcalls -mtext-section-literals -falign-functions=4 -ffunction-sections -fdata-sections -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=030100 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DARCH_ESP -DFAMILY_ESP -DMCU_ESP8266 -DBOARD_ESP12E -DVERSION=esp32 -DICACHE_FLASH -D__STRICT_ANSI__ -D__ets__ -U__STRICT_ANSI__ -I.pioenvs/esp12e/SimbaFramework/src -I/Users/wwx/.platformio/packages/framework-simba/src -I.pioenvs/esp12e/SimbaFramework/3pp/ESP8266_RTOS_SDK/include -I/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include -I.pioenvs/esp12e/SimbaFramework/build/esp12e -I/Users/wwx/.platformio/packages/framework-simba/build/esp12e -I.pioenvs/esp12e/SimbaFramework/3pp/ESP8266_RTOS_SDK/include/espressif/esp8266 -I/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/espressif/esp8266 -I/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/espressif/esp8266 -I/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/espressif/esp8266 -I.pioenvs/esp12e/SimbaFramework/3pp/spif [... cut ...] 8266_RTOS_SDK/include/espressif -I/Users/wwx/.platformio/packages/framework-simba/3pp/ESP8266_RTOS_SDK/include/espressif -I.pioenvs/esp12e/SimbaFramework/src/boards/esp12e -I/Users/wwx/.platformio/packages/framework-simba/src/boards/esp12e -I/Users/wwx/.platformio/packages/framework-simba/src/boards/esp12e -I.pioenvs/esp12e/SimbaFramework/src/kernel/ports/esp/gnu -I/Users/wwx/.platformio/packages/framework-simba/src/kernel/ports/esp/gnu -I.pioenvs/esp12e/SimbaFramework/3pp/ESP8266_RTOS_SDK/inclu

eerimoq, 2016-10-23:
Windows PC? I've only tested on Linux and sometimes the toolchain behaves differently on Windows.

eerimoq, 2016-10-23:
I reverted the platformio.sconsscript changes and pushed to master. Can you test it?

wuwx, 2016-10-23:
oh, it's work now

eerimoq, 2016-10-23:
:P

eerimoq, 2016-10-23:
There is a new version of platformio.sconscript on master now. Tested PlatformIO on windows and it works for me. I'll leave this issue open until you confirm that it works for you.

wuwx, 2016-10-23:
i try to find out why system broken

if the .c code call atoi func, then the system will be broken, use latest master branch at commit aa8d2740a6802cc3252391a7738417712f437087

main.c

```
#include "simba.h"
int main()
{
    sys_start();
    std_printf("%d", atoi("1"));
    return 0;
}
```

platformio.ini

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter, extra scripting
;   Upload options: custom port, speed and extra flags
;   Library options: dependencies, extra library storages
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/en/stable/projectconf.html

[env:esp12e]
platform = espressif8266
framework = simba
board = esp12e

```

eerimoq, 2016-10-23:
Does it work with version 10.0.0 of platformio.sconscript, or has it always been like that? I've not used atoi() on the esp.

wuwx, 2016-10-23:
at commit 5fb8034af97bd7dbc66aa3f84cf77751cd92d231
it's work fine.

and ask a question, how to generate esp8266 ld script ?

eerimoq, 2016-10-23:
The ld script is not generated. ``platformio.sconscript`` is generated with ``make platformio`` in the root folder of Simba.

wuwx, 2016-10-23:
it's complex find out solution ...

eerimoq, 2016-10-23:
Try to move ``"minic"`` in ``platformio.sconscript`` from the second element in the list to the last in the lib array. It seem to solve the problem.

eerimoq, 2016-10-23:
I pushed the change. Let me know if there is still a problem.

wuwx, 2016-10-23:
atoi function works
but still has other problem
do you has any online talk tool ?
some like gtalk, qq, or other ?

wuwx, 2016-10-23:
```
#include "simba.h"
#include "esp_wifi.h"

static void heater_network_callback(System_Event_t *event)
{
}

int main()
{
    sys_start();
    std_printf("%d", atoi("1"));
    wifi_set_event_handler_cb(heater_network_callback);
    return 0;
}
```

wifi_set_event_handler_cb function broken ..

eerimoq, 2016-10-23:
Let's try gtalk. How can I contact you?

wuwx, 2016-10-23:
wuweixin@gmail.com

closed 2016-10-23 by eerimoq