eerimoq/simba #61: Linux board make error after commit >= 156e0aa

wuwx, 2016-12-01:
```
>>> app: hello_world, board: linux <<<

CC /home/wuwx/tmp/simba/src/inet/network_interface/slip.c
In file included from /home/wuwx/tmp/simba/3pp/lwip-1.4.1/src/include/lwip/arch.h:43,
                 from /home/wuwx/tmp/simba/3pp/lwip-1.4.1/src/include/lwip/debug.h:35,
                 from /home/wuwx/tmp/simba/3pp/lwip-1.4.1/src/include/lwip/opt.h:46,
                 from /home/wuwx/tmp/simba/3pp/lwip-1.4.1/src/include/lwip/init.h:35,
                 from /home/wuwx/tmp/simba/src/inet/network_interface/slip.c:43:
/home/wuwx/tmp/simba/src/inet/arch/cc.h:39: error: redefinition of typedef 'u8_t'
/home/wuwx/tmp/simba/src/filesystems/spiffs/config.h:252: note: previous declaration of 'u8_t' was here
/home/wuwx/tmp/simba/src/inet/arch/cc.h:40: error: redefinition of typedef 's8_t'
/home/wuwx/tmp/simba/src/filesystems/spiffs/config.h:255: note: previous declaration of 's8_t' was here
/home/wuwx/tmp/simba/src/inet/arch/cc.h:41: error: redefinition of typedef 'u16_t'
/home/wuwx/tmp/simba/src/filesystems/spiffs/config.h:253: note: previous declaration of 'u16_t' was here
/home/wuwx/tmp/simba/src/inet/arch/cc.h:42: error: redefinition of typedef 's16_t'
/home/wuwx/tmp/simba/src/filesystems/spiffs/config.h:256: note: previous declaration of 's16_t' was here
/home/wuwx/tmp/simba/src/inet/arch/cc.h:43: error: redefinition of typedef 'u32_t'
/home/wuwx/tmp/simba/src/filesystems/spiffs/config.h:254: note: previous declaration of 'u32_t' was here
/home/wuwx/tmp/simba/src/inet/arch/cc.h:44: error: redefinition of typedef 's32_t'
/home/wuwx/tmp/simba/src/filesystems/spiffs/config.h:257: note: previous declaration of 's32_t' was here
make[1]: *** [build/linux/obj/home/wuwx/tmp/simba/src/inet/network_interface/slip.o] Error 1
make: *** [all] Error 2

```

may because this:

https://github.com/eerimoq/simba/commit/156e0aa#diff-647781977468bf0c06fd2944e09e9380

wuwx, 2016-12-01:
if gcc > 4.8 (centos 7), it works fine
if gcc < 4.8 (centos 6), it's broken

eerimoq, 2016-12-01:
Fixed in 7bc870418796dada4f196f5dbd402ce5d8a04b7d. The typedef is in one place now, no duplication.

wuwx, 2016-12-02:
oh, yeah!
it's fine.

closed 2016-12-02 by wuwx

rahuldeo2047, 2017-10-04:
@eerimoq : 
Question: 

In case using platformIO 
what should be in platformIO.ini? Or when to look for these parameters.
e.g.:

[env:linux]
platform = linux ?? 
framework = simba
board = linux 

If I put linux_x86_64 or native as per platformIO doc, it says invalid ID linux for board option.

Any help here?

eerimoq, 2017-10-04:
The linux build is only supported by the Make base build framework called [Simba build system](http://simba-os.readthedocs.io/en/latest/getting-started.html#simba-simba-build-system).

rahuldeo2047, 2017-10-04:
OK