eerimoq/simba #62: request implement ssl/tls layer

wuwx, 2016-12-01:
improve data security

eerimoq, 2016-12-01:
I created a file ``inet/ssl.h``, with a few functions to initialize, read and write data over SSL/TLS. See commit c707e42fd046344af81a55620484c78ab9373e5d.

I don't have any experiance using SSL/TLS libraries. Seems like mbedTLS is popular. What alternatives are free and open source? Feel free to make changes to ssl.h and add more files for SSL.

I guess the ssl instance should be a channel, since it is a data transport protocol (with some security on top).

eerimoq, 2016-12-02:
Added bais client and server examples in a [test suite](https://github.com/eerimoq/simba/blob/master/tst/inet/ssl/main.c) to understand how to use the SSL module interface.

eerimoq, 2016-12-02:
The plan is to design the interface as the Python [ssl module](https://docs.python.org/3/library/ssl.html) interface.

wuwx, 2016-12-03:
https://github.com/espressif/esp8266-rtos-sample-code/tree/master/04Protocal/SSL_Demo

esp8266 rtos ssl demo project

eerimoq, 2017-01-01:
I've successfully setup a SSL server side socket between a ESP32 and my PC! Needs tuning of memory usage and, most importantly, a valid entropy source implementation to ensure(?) secure communication.

See the [example](https://github.com/eerimoq/simba/blob/master/examples/ssl_server/main.c) for usage.

eerimoq, 2017-01-03:
The first version is pushed and released. Let me know if there are issues or features that are missing. Only server side is implemented.

eerimoq, 2017-01-06:
Both client and server side has been implemented and seems to work. Ther eare still functionality missing that should be added to make the security better, like verify the hostnames.

closed 2017-01-06 by eerimoq