eerimoq/simba #64: ESP32: I2C driver.

eerimoq, 2016-12-02:
Port the I2C driver to ESP32.

cicamargoba, 2016-12-13:
Hello Erik
I want to port ADC to simba, this is possible using the espressif sdk (I mean espressif drivers) ?

eerimoq, 2016-12-13:
The ESP8266 ADC driver port calls the Espressif SDK driver, and it works for single sample convertions. I've not seen the ESP32 Espressif driver API, so I don't have a good answer, but it will probably work :)

cicamargoba, 2016-12-14:
> Thanks for your answer
>
> I'll use the esp8266 driver as a guide
>
>
> Carlos

Hello Erik
One question:

I understand that I need to create a driver in
simba/src/drivers/ports/esp32/  right?

How can I use this new driver on microptthon?

Best regards

Carlos

eerimoq, 2016-12-15:
As usual when someone asks about a new feature I cannot resist implementing it myself. =) I started reading the technical reference, the data sheet and some header files of the IDF, and started the ADC port to ESP32. No success reading analog pins so far, but unless there is a lot of non-documented features required it should be ready in a few days.

For Pumbaa (I guess you are intrested in this MicroPython port), the ADC driver wrapper is already in place. Just have to set CONFIG_PUMBAA_CLASS_ADC to 1 and add class_adc.c to the list of files to build for ESP32. And of course, update the simba submodule to a version including the ADC driver port.

cicamargoba, 2016-12-15:
Very good for me :)
I'll be waiting for updates
Thanks a lot.

Carlos

eerimoq, 2016-12-15:
Moving ADC discussion to #65.

eerimoq, 2017-02-20:
Postponed.

closed 2017-02-20 by eerimoq

LupascuAndrei, 2017-05-31:
Hello there :D 
Any chance the i2c port will be done soon? 
Thank you in advance.

eerimoq, 2017-05-31:
Hi,

I don't have the energy to looks into it at the moment. Too much going on already. =)