eerimoq/simba #65: ESP32: ADC driver.

eerimoq, 2016-12-13:
Port the ADC driver to ESP32.

eerimoq, 2016-12-17:
An ADC port using the ESP-IDF has been implemented and pushed. Only ADC1 works, that is, GPIO 32-35, 36 and 39. ADC2 is not yet implemented in the ESP-IDF.

eerimoq, 2016-12-17:
@cicamargoba Let me know if you find any issues with the port.

closed 2016-12-17 by eerimoq

cicamargoba, 2016-12-28:
Hello Erik

Thanks for this port,

I'm trying to compile the simba/analog_read example (downloaded from pumbaa
git repository ) and the process fails with:

>>> app: analog_read, board: nano32 <<<

CC main.c
main.c: In function 'main':
main.c:36:31: error: storage size of 'pin' isn't known
     struct analog_input_pin_t pin;
                               ^
main.c:39:5: error: implicit declaration of function
'analog_input_pin_module_init' [-Werror=implicit-function-declaration]
     analog_input_pin_module_init();

Best Regards

Carlos

eerimoq, 2016-12-28:
Hi Carlos,

the [analog input pin header file](https://github.com/eerimoq/simba/blob/master/src/drivers/analog_input_pin.h) was not included in [simba.h](https://github.com/eerimoq/simba/blob/master/src/simba.h). I added it just now in commit ae08036316545aff18e2ba168a1a2a966f612815.

cicamargoba, 2016-12-28:
Thank Erik

The example compile fine, tomorrow I'll test on my board (mid night here)
There is any example using ADC/DAC on pumpaa?
Best regards

cicamargoba, 2016-12-28:
Hello Erik

If I make a pull on simba repository (git clone
https://github.com/eerimoq/simba.git) (inside pumbaa) the compilation fails:
[cain@aldebaran-mac pumbaa]$ cd examples/blink/
[cain@aldebaran-mac blink]$ make BOARD=nano32

>>> app: blink, board: nano32 <<<

make prepare
make[1]: Entering directory
`/home/cain/Embedded/ESP32/pumbaa/examples/blink'
make[1]: Nothing to be done for `prepare'.
make[1]: Leaving directory `/home/cain/Embedded/ESP32/pumbaa/examples/blink'
make generate
make[1]: Entering directory
`/home/cain/Embedded/ESP32/pumbaa/examples/blink'
make[1]: *** No rule to make target
`/home/cain/Embedded/ESP32/pumbaa/simba/3pp/mbedtls/library/aes.c', needed
by `build/nano32/genhdr/qstr.i.last'.  Stop.
make[1]: Leaving directory `/home/cain/Embedded/ESP32/pumbaa/examples/blink'
make: *** [all] Error 2

Carlos

cicamargoba, 2016-12-28:
When I trying to compile a simba example:

[cain@aldebaran-mac simba]$ cd examples/analog_read/
[cain@aldebaran-mac analog_read]$ make BOARD=nano32

>>> app: analog_read, board: nano32 <<<

make prepare
make[1]: Entering directory
`/home/cain/Embedded/ESP32/pumbaa/simba/examples/analog_read'
make[1]: Nothing to be done for `prepare'.
make[1]: Leaving directory
`/home/cain/Embedded/ESP32/pumbaa/simba/examples/analog_read'
make generate
make[1]: Entering directory
`/home/cain/Embedded/ESP32/pumbaa/simba/examples/analog_read'
make[1]: Nothing to be done for `generate'.
make[1]: Leaving directory
`/home/cain/Embedded/ESP32/pumbaa/simba/examples/analog_read'
make build
make[1]: Entering directory
`/home/cain/Embedded/ESP32/pumbaa/simba/examples/analog_read'
CC main.c
mkdir -p
build/nano32/obj/home/cain/Embedded/ESP32/pumbaa/simba/examples/analog_read
mkdir -p
build/nano32/deps/home/cain/Embedded/ESP32/pumbaa/simba/examples/analog_read
mkdir -p build/nano32/gen
xtensa-esp32-elf-gcc -I. -I/home/cain/Embedded/ESP32/pumbaa/simba/src
-Ibuild/nano32 -I/home/cain/Embedded/ESP32/pumbaa/simba/src/boards/nano32
-I/home/cain/Embedded/ESP32/pumbaa/simba/src/mcus/esp32
-I/home/cain/Embedded/ESP32/pumbaa/simba/3pp/esp32/inc/driver/include
-I/home/cain/Embedded/ESP32/pumbaa/simba/3pp/esp32/inc/esp32/include
-I/home/cain/Embedded/ESP32/pumbaa/simba/3pp/esp32/inc/newlib/include
-I/home/cain/Embedded/ESP32/pumbaa/simba/3pp/esp32/inc/freertos/include
-I/home/cain/Embedded/ESP32/pumbaa/simba/3pp/esp32/inc/nvs_flash/include
-I/home/cain/Embedded/ESP32/pumbaa/simba/3pp/esp32/inc/spi_flash/include
-I/home/cain/Embedded/ESP32/pumbaa/simba/3pp/esp32/inc/tcpip_adapter/include
-I/home/cain/Embedded/ESP32/pumbaa/simba/3pp/esp32/inc/lwip/include/lwip
-I/home/cain/Embedded/ESP32/pumbaa/simba/3pp/esp32/inc/lwip/include/lwip/port
-I/home/cain/Embedded/ESP32/pumbaa/simba/src
-I/home/cain/Embedded/ESP32/pumbaa/simba/3pp/compat
-I/home/cain/Embedded/ESP32/pumbaa/simba/src/drivers/ports/esp32
-I/home/cain/Embedded/ESP32/pumbaa/simba/3pp/mbedtls/include
-I/home/cain/Embedded/ESP32/pumbaa/simba/src/kernel/ports/esp32/gnu
-I/home/cain/Embedded/ESP32/pumbaa/simba/3pp/spiffs-0.3.5/src
-I/home/cain/Embedded/ESP32/pumbaa/simba/3pp/atto
-I/home/cain/Embedded/ESP32/pumbaa/simba/3pp/atto/curses -DARCH_ESP32
-DFAMILY_ESP32 -DMCU_ESP32 -DBOARD_NANO32 -DVERSION=master
-DMBEDTLS_USER_CONFIG_FILE="\"mbedtls/user_config.h\"" -DF_CPU=240000000UL
-Os -Werror -nostdlib -mlongcalls -ffunction-sections -fdata-sections -c
-Wall -funsigned-bitfields -std=gnu99 -Wno-error=unused-variable -o
build/nano32/obj/home/cain/Embedded/ESP32/pumbaa/simba/examples/analog_read/main.o
main.c
In file included from
/home/cain/Embedded/ESP32/pumbaa/simba/src/simba.h:70:0,
                 from main.c:31:
/home/cain/Embedded/ESP32/pumbaa/simba/src/kernel/ports/esp32/gnu/compiler.h:34:22:
fatal error: esp_attr.h: No such file or directory
 #include "esp_attr.h"
                      ^
compilation terminated.
make[1]: ***
[build/nano32/obj/home/cain/Embedded/ESP32/pumbaa/simba/examples/analog_read/main.o]
Error 1
make[1]: Leaving directory
`/home/cain/Embedded/ESP32/pumbaa/simba/examples/analog_read'
make: *** [all] Error 2

eerimoq, 2016-12-28:
``source setup.sh`` in the pumbaa root will setup the pumbaa development environment. This setup script does not setup the environment to build simba examples in the simba submodule. I have a separate simba repository for simba development, where I run ``source setup.sh`` in the root to setup the development environment.