eerimoq/simba #70: Looking for a websocket+web server example

josephwinston, 2016-12-20:
I've found ```inet/http_websocket_server/main.c``` and would like to integrate the websocket code into the http_server example you have already written. My attempt here is based on ```http_server``` <https://gist.github.com/josephwinston/fc99563cfc70967973200c81661c3d3e>

```curl -v``` reports:

```
$ curl -v http://192.168.1.251:8000/stream
*   Trying 192.168.1.251...
* Connected to 192.168.1.251 (192.168.1.251) port 8000 (#0)
> GET /stream HTTP/1.1
> Host: 192.168.1.251:8000
> User-Agent: curl/7.49.0
> Accept: */*
>
* Empty reply from server
* Connection #0 to host 192.168.1.251 left intact
curl: (52) Empty reply from server
```

Obviously, I'm not doing something correctly.  Is there an example that I can use that shows the integration of a websockets into a websert?

josephwinston, 2016-12-20:
One issue was that I did not correctly initialize the websocket.  The updated example is here <https://gist.github.com/josephwinston/bad243712d8ba6fc91b80176dfd5a28b>

```curl -v``` now responds with:

```
$ curl -v http://192.168.1.251:8000/stream
*   Trying 192.168.1.251...
* Connected to 192.168.1.251 (192.168.1.251) port 8000 (#0)
> GET /stream HTTP/1.1
> Host: 192.168.1.251:8000
> User-Agent: curl/7.49.0
> Accept: */*
>
< HTTP/1.1 101 Switching Protocols
< Connection: Upgrade
< Sec-WebSocket-Accept: Kfh9QIsMVZcl6xEPYxPHzW8SZ8w=
* Connection #0 to host 192.168.1.251 left intact
��foo
```

Open questions include:
1. Do I perform the ```http_websocket_server_init``` on each connection?
1. Does the ```http_websocket_server_handshake``` get called every time the endpoint is called?
1. After the ```http_websocket_server_write```, is there anything else that I need to handle?

eerimoq, 2016-12-20:
This test case acts as an example of how the intended usage: https://github.com/eerimoq/simba/blob/master/tst/inet/http_server/main.c#L115-L151

In short, the connection thread that handles the client HTTP websocket request enters a loop communicating with the client. When the websocket is closed the connection thread is once again available fto handle other requests.

closed 2016-12-20 by josephwinston