eerimoq/simba #71: ESP32 CAN Bit Timing

ThomasBarth, 2017-01-03:
Hi!
There is a bug in https://github.com/eerimoq/simba/blob/master/src/drivers/ports/esp32/can_port.h where 1MBit and 500kBits have the same settings.

I calculated all the settings with an online generator:
    	//Values based on http://www.esacademy.com/en/library/calculators/sja1000-timing-calculator.html with 80Mhz Clock
		case CAN_SPEED_1000KBPS:
			MODULE_CAN->BTR0.B.BRP		=0x4;
			MODULE_CAN->BTR0.B.SJW		=0x1;
			MODULE_CAN->BTR1.B.TSEG1	=0x4;
			MODULE_CAN->BTR1.B.TSEG2	=0x1;
			break;

		case CAN_SPEED_500KBPS:
			MODULE_CAN->BTR0.B.BRP		=0x4;
			MODULE_CAN->BTR0.B.SJW		=0x1;
			MODULE_CAN->BTR1.B.TSEG1	=0xc;
			MODULE_CAN->BTR1.B.TSEG2	=0x1;
			break;

		case CAN_SPEED_250KBPS:
			MODULE_CAN->BTR0.B.BRP		=0x9;
			MODULE_CAN->BTR0.B.SJW		=0x1;
			MODULE_CAN->BTR1.B.TSEG1	=0xc;
			MODULE_CAN->BTR1.B.TSEG2	=0x1;
			break;

eerimoq, 2017-01-03:
Hi, thanks for reporting the issue. The settings for 1MBit and 250kBit (not 500kBit) are commented out because they are not correct. I copied them from the Arduino Due port. The 500kBit settings worked when I tested using an Arduino Due and a ESP32. However, it was more trial an error than calculating the settings.

Did you configure the SJA1000 on the ESP32 with the suggested values from the online calculator? If not, can you? When it works, please submit a pull request with the changes. Just fork this repository, make your changes and submit the pull request.

ThomasBarth, 2017-01-03:
Yes I used and tested the suggested values from the calculator. They work fine.

eerimoq, 2017-01-03:
Merged. Will make a test myself as well for 500 kBit.

eerimoq, 2017-01-03:
Okay, so with the latest ESP IDF gpio 32 and 33 are probably allocated for the ULP by default, at least the TX pin does not work anymore. I changed the CAN pins to 16 and 17, and 500 kBit works very well with your settings, just as expected.

closed 2017-01-03 by eerimoq

ThomasBarth, 2017-01-04:
Yes, I ran into the same problem. I use Pin 4 and 5.