eerimoq/simba #75: Direct write to output port register

pctj101, 2017-02-23:
Is it possible to directly write to PORTB of an Atmega?  

The goal being to change 2 digital pin-outs at the "exact same time".

eerimoq, 2017-02-23:
Not with the current pin module API. It only implements writing to individual pins. Submit a pull request if you want to add it as part of Simba. All help is appreciated =)

Here is how to write to single pins on AVR:

https://github.com/eerimoq/simba/blob/master/src/drivers/ports/avr/pin_port.h#L66-L78

pctj101, 2017-02-24:
Thanks for the info.  Let me see how I can help.

I'm stuck at a few points.  I see how you're using the mask to toggle bits for *PORT, which seems to come from https://github.com/eerimoq/simba/blob/97f2154307a3d462c6710da50a9489e5a5ad3f58/src/drivers/ports/avr/pin_port.h#L36

That seems to get set up here:
https://github.com/eerimoq/simba/blob/6ca9662b3bcd8b013d45705d8c44ce2a8ae71cb4/src/mcus/atmega328p/mcu.c

But I'm having trouble understanding where PINA/PINB/PINC/_BC/PINB# are defined.  Is it not in this source tree?

Correct me if I'm wrong, but It seems line the "PINX" here is what I'm used to seeing as "PORTX".  I'm a bit lacking in history here, but is there a reason why simba's PINX refers to "several pins"?

Once I understand that I think I can do the bit fiddling.  

I'm on OSX PlatformIO.  Is there a way to make it reference a dev branch so that I can dev/test this new "non-published" code?

pctj101, 2017-02-24:
Ok, PIN* for read, PORT* for write

pctj101, 2017-02-24:
So it looks like in a pinch I can simply directly bit-bang PORT*?  Because in a way, I feel that most people don't really have the timing requirement I have to flip 2 bits at once.  And if they do, an API isn't exactly the most helpful thing since which pins to flip are so "varied"

eerimoq, 2017-02-24:
All registers and low level difinitions like PINA/B/C/_BV are defined in the AVR C runtime library http://savannah.nongnu.org/projects/avr-libc/. It is installed in ``/usr/lib/avr`` on my PC.

Here is how to make changes to Simba:

1. Fork this reposiory here on GitHub.
2. Rename ``~/.platformio/packages/framework-simba/`` to ``~/.platformio/packages/framework-simba.orig/`` to keep a copy of the original folder.
3. Run ``git clone https://github.com/pctj101/simba framework-simba`` in ``~/.platformio/packages``.
4. Make your changes to the code in the newly cloned repository.
5. Use platformio to build and upload as usual.

pctj101, 2017-02-24:
Wow thanks, I'll give it a shot!

eerimoq, 2017-02-24:
=)

pctj101, 2017-02-24:
So I played around a bit and it really makes more sense to go directly at the PORTX for multiple bit changes.  I don't think there's a super good way of "abstracting this" across multiple MCU's or multiple PORTX's (at least it doesn't seem clear at this point).

So I'm going to close this for now.  If you think there's something I missed let me know.  I followed your instructions and I have the git repo in my platformio package now, so it shouldn't be a big deal either way.

Thanks again!

closed 2017-02-24 by pctj101