eerimoq/simba #78: PCINT (Pin Change Interrupts)

pctj101, 2017-02-24:
It looks like the exti module only supports exti_device_t based pins.  Which is entirely understandable.

Is there a module I missed that handles Pin Change Interrupts?
http://playground.arduino.cc/Main/PinChangeInterrupt
http://playground.arduino.cc//Main/PinChangeIntExample

Is it something that should be added to simba?

pctj101, 2017-02-24:
Or should I just simply use:
```
 ISR (PCINT1_vect) // handle pin change interrupt for A0 to A5 here
```

And be happy with it?  Is there something "more simba-like" that I'm supposed to do to be more hardware agnostic?

eerimoq, 2017-02-24:
How about adding a new trigger in the exti module ``EXTI_TRIGGER_CHANGE``. Pretty much the same as ``EXTI_TRIGGER_BOTH_EDGES``, but will configure the change pin registers. I don't know how often MCU:s have pin change interrupts, but it sounds like more MCU:s could use this new trigger type. The only drawback I can see is higher code and data size, but we could make it configurable. Not a big deal.

pctj101, 2017-02-24:
Ah that's interesting.  Let me see if I can figure out how to manually configure those pin registers.  If I can do that, then I think I could add those functions you mentioned.  Obviously... I don't entirely know what I'm doing :)

eerimoq, 2017-02-24:
On second thought, I think it's better to add a new module called ``pcint`` and take a ``pin_device_t`` as pin to register the interrupt to.

The interface will be similar to the ``exti`` module, but taking a ``pin_device_t`` instead of a ``exti_device_t``.

It's always hard to make the right decision first try, have to start implementing and see what problems one runs into =) Maybe we will end up to remove the ``exti_device_t`` and pass a ``pin_device_t`` to the ``exti`` module. Was a long time since I designed the ``exti`` module so can't really remember why it looks like it does =)

pctj101, 2017-02-24:
Hahaha fair enough and so true.  For the PinChange Interrupts it looks like in the ATmega world we're looking at a few things.  

SREG - Enable Interrupts
PCICR - Enable PinChange Interrupts for a certain Register (bank of 8 pins)
PCMSK# - Which pins in the bank qualify 
PCIFR - The flag which is raised on interrupts.

With that said, it looks much more involved than the exti_device_t would allow naturally so a different "something" might be a good idea.

Looks like the configuration to launch the interrupt on pin change works, but for me so far... only once :(

Time to learn more... :)

eerimoq, 2017-02-24:
:P

pctj101, 2017-02-24:
Okay things are fine.  I had to re-teach myself to use | and not || for bit operations. :)

pctj101, 2017-02-24:
So there's a few options to store all that metadata:

1) Modify pin_device_t or exti_device_t to be aware of PCICR, PCIFR and PCMSK#/PCINT bits
  * Downside, maybe other MCUs don't want this metadata in the same struct
https://github.com/eerimoq/simba/blob/97f2154307a3d462c6710da50a9489e5a5ad3f58/src/drivers/ports/avr/pin_port.h
https://github.com/eerimoq/simba/blob/97f2154307a3d462c6710da50a9489e5a5ad3f58/src/drivers/ports/avr/exti_port.h

2) Create pin_change_device_t which is like exti_device_t (stores a pointer to the original pin_device_t) and add metadata in a nicely isolated area

3) Make some nasty function just for AVR and mix with: (looks like such a bad idea)
https://github.com/eerimoq/simba/blob/97f2154307a3d462c6710da50a9489e5a5ad3f58/src/drivers/ports/avr/pin_port.i
https://github.com/eerimoq/simba/blob/97f2154307a3d462c6710da50a9489e5a5ad3f58/src/boards/arduino_nano/board.c

As a side note, looks like we need to create a bunch of ISR##() handles to call self_p->on_interrupt(self_p->arg_p);         
https://github.com/eerimoq/simba/blob/97f2154307a3d462c6710da50a9489e5a5ad3f58/src/drivers/ports/avr/exti_port.i

These PC ISRs might be yet another reason to go with a pin_change_device_t (is there a shorter name we would be happy with?)

eerimoq, 2017-02-24:
Let's start by creating a new device struct called ``struct pcint_device_t`` and put the metadata in it. Call the module ``pcint`` as well. Most drivers are abbreviations already, so it is ok.

Yeah, create a bunch of ISR##() calling the callback as you wrote, very much like the ``exti`` module.

pctj101, 2017-02-24:
Okay will give a go at it another day.  Thanks a bunch!

pctj101, 2017-03-28:
Hey a quick update.  The project I was working on didn't work out on this hardware, so I need to take care of work on a different board before I can look into this topic again.

eerimoq, 2017-07-19:
Implemented, but not yet tested. Will be tested soon.

closed 2017-07-19 by eerimoq