eerimoq/simba #81: Doc builds fails - xtensa-lx106-elf-gcc: command not found

simonlbn, 2017-02-25:
When I try to build the documentation by using "make doc" (as doc/README.rst says) in the project root directory it fails, complaining it cannot find xtensa-lx106-elf-gcc. I haven't found any Ubuntu or other package for xtensa-lx106-elf-gcc - it seems to come commonly from the esp8266 build toolchain.

So, is "make doc" expected to work, and if so - what am I missing on my devel system?

Note that if I just do "cd doc && make" I get the main documentation built successfully.

Full output:

simon@simon-ubuntu:~/proj/simba-doc$ make doc
bin/dbgen.py > database.json
bin/docgen.py database.json

>>> app: minimal_configuration, board: esp01 <<<

CC main.c
/bin/bash: xtensa-lx106-elf-gcc: command not found
/home/simon/proj/simba-doc/make/app.mk:227: recipe for target 'build/esp01/obj/home/simon/proj/simba-doc/examples/minimal-configuration/main.o' failed
make[2]: *** [build/esp01/obj/home/simon/proj/simba-doc/examples/minimal-configuration/main.o] Error 127
/home/simon/proj/simba-doc/make/app.mk:105: recipe for target 'all' failed
make[1]: *** [all] Error 2
Traceback (most recent call last):
  File "bin/docgen.py", line 279, in <module>
    main()
  File "bin/docgen.py", line 273, in main
    boards_generate(database)
  File "bin/docgen.py", line 169, in boards_generate
    'all'])
  File "/usr/lib/python2.7/subprocess.py", line 541, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-s', '-C', 'examples/minimal-configuration', u'BOARD=esp01', 'all']' returned non-zero exit status 2
Makefile:527: recipe for target 'doc' failed
make: *** [doc] Error 1

eerimoq, 2017-02-25:
Building the documentation from the repository root will build minimal and default configuration applications for all targets. Memory footprint statistics documentation pages are generated from the built binaries.

There is a long one-liner on the getting started page where esp-open-sdk is built and esp32 is downloaded and unpacked.

http://simba-os.readthedocs.io/en/latest/getting-started.html#simba-simba-build-system

The generated documentation is not that important except when preparing a release. I think I should change the script to print a warning and continue instead of bailing out if a toolchain is missing.

eerimoq, 2017-02-25:
I pushed a fix that does not about on compilation failure. Should solve your problem.

simonlbn, 2017-02-25:
Thanks! I will try that.

I had built with platformio before and for some reason didn't think of reading reading the notes for native Simba build for the doc build.

I'm also started installing the normal build tools now, but that seems to take a bit (running in a VM probably doesn't help).

eerimoq, 2017-02-25:
Yeah, it takes quite a while to build a toolchain. Takes my machine about 20 minutes.

simonlbn, 2017-02-25:
Building doc from root directory works (boards fail of course but rest succeeds) now, thanks!

closed 2017-02-25 by simonlbn