eerimoq/simba #82: OneWire Reads corrupted by interrupts

pctj101, 2017-02-27:
It appears that if OneWire devices are being read from (and written to (probably) for that matter) while another interrupt executes that timing may be disturbed (sometimes).

Thus the resulting data reads may contain errors.

Currently checksums are not being calculated/checked.  If they were, we could more easily detect these read errors.

May be able to save work if CRC polynomial is the same as other CRC already in simba.  Otherwise, may want to isolate to OneWire.

Also for consideration, slower but less code CRC, or faster but more memory hungry CRC lookup tables.

http://forum.arduino.cc/index.php?topic=37648.0
https://www.maximintegrated.com/en/app-notes/index.mvp/id/27

eerimoq, 2017-02-27:
Which board are you using? I want to test on the same board.
Which OneWire device are you using?

The OneWire communication has indeed very high requirements on timing, and since it's implemented in software it disables interrupts while communicating over the bus. I've mainly tested the OneWire driver on AVR with DS18B20, and it works fine. The function ``time_busy_wait_us()`` may not sleep for the correct number of microseconds on some targets. Might be related to this error.

I'll have a look and probably add CRC calculations.

pctj101, 2017-02-27:
Yeah I'm working with an Arduino Uno & DS18B20.

eerimoq, 2017-02-27:
Ohh okay, strange, last time I used that MCU the OneWire worked. However, my applications are usually testing one driver at a time and does not stress the system.

Does OneWire work if you remove everything else from your application? Test something like this: https://github.com/eerimoq/simba/blob/master/examples/ds18b20/main.c

pctj101, 2017-02-27:
I didn't get a chance to run your example, but I know that if I don't fire the interrupts, the readings are quite stable :)

eerimoq, 2017-02-27:
Pushed a DS18B20 driver with CRC calculation on the crc_8 branch. I have *not* tested it on the UNO, will do that tonight. Give it a try if you want.

https://github.com/eerimoq/simba/commit/5ceba7ef2c5fa340839020baa75513750e9c9573

pctj101, 2017-02-27:
Hi, tested your branch.  Works well.  

During high interrupt periods, I either get 0 detected OWI devices or CRC errors (-1) reported back.  That works just fine for me.

eerimoq, 2017-02-27:
Just pushed the CRC check to master. Closing this issue knowing that the root cause has not been solved.

closed 2017-02-27 by eerimoq

pctj101, 2017-02-28:
Thanks.  By the way, do you have any ideas how you would be able to attack this issue? Or do you think it's kind of a "lost cause"?

I think to a certain extent it's really hard unless you have a processor so fast it doesn't matter.

But also perhaps it is possible to eek out just a little more with a different structure?

Any gut feelings?

eerimoq, 2017-02-28:
It's really hard to say without reproducing the problem and troubleshoot it. Interrupts are disabled around time critical blocks in the ``owi`` module, but maybe I missed one. I don't know. ``sys_lock()`` disabled interrupts, while ``sys_unlock()`` enabled them. There may also be something in the OWI specification I'm misinterpreted.

eerimoq, 2017-02-28:
Do you have an application I could use to reproduce the issue?

pctj101, 2017-03-01:
I'll get back to you in about a week.  :)

eerimoq, 2017-03-01:
Ok!

pctj101, 2017-03-11:
@eerimoq I wanted to let you know that after streamlining the interrupt handler I no longer "notice" the one wire read errors.  Granted sure there might be some CRC errors here and there, but not enough for me to raise an issue at this time.  "Solid enough" for me.

Thanks a bunch for the support!