eerimoq/simba #88: Doc build fails for spc56ddiscovery

simonlbn, 2017-03-18:
Hi,

I can't build the doc for spc56ddiscovery due to missing "powerpc-eabivle-gcc". I looked a bit around but couldn't find where to get this compiler.

Log output:

```
>>> app: minimal_configuration, board: spc56ddiscovery <<<

CC main.c
/bin/bash: powerpc-eabivle-gcc: command not found
/home/simon/proj/simba/master/make/app.mk:238: recipe for target 'build/spc56ddiscovery/obj/home/simon/proj/simba/master/examples/minimal-configuration/main.o' failed
make[2]: *** [build/spc56ddiscovery/obj/home/simon/proj/simba/master/examples/minimal-configuration/main.o] Error 127
/home/simon/proj/simba/master/make/app.mk:114: recipe for target 'all' failed
make[1]: *** [all] Error 2
Failed to generate memory footprint data for board spc56ddiscovery. Skipping board.
```

eerimoq, 2017-03-18:
See http://simba-os.readthedocs.io/en/latest/boards/spc56ddiscovery.html#toolchain.

simonlbn, 2017-03-18:
Ah, I should have looked there - I just assumed it just hadn't been added to the quickstart yet, but looking at it I can see why it's not done by default.

Thanks!

closed 2017-03-18 by eerimoq