eerimoq/simba #95: ESP8266 - specifying station BSSID to connect to

eimiss, 2017-04-13:
Hello,
[noob alert]
Seems like Simba doesn't have an option to specify WiFi station BSSID to connect to. This is important when connecting to multi-AP WiFi network because SDK is too dumb to figure out which AP to connect to on its own.
Espressiv SDK does support this via the station_config struct. I'm wondering whether it's planned to add this into Simba, or am I just missing something?

eerimoq, 2017-04-13:
Is this what you are looking for?

https://github.com/eerimoq/simba/blob/709c3756bb417ceabfd1a5ab715da4e72ac3a101/src/drivers/esp_wifi/station.h#L49-L67

eimiss, 2017-04-13:
Thank you for taking your time to reply.
What I mean is the access point mac - specifying that while connecting to an AP.

In Espressive SDK we can use the station_config struct to define not only the SSID but also BSSID.
```c
//esp_sta.h
struct station_config {
    uint8 ssid[32];         /**< SSID of target AP*/
    uint8 password[64];     /**< password of target AP*/
    uint8 bssid_set;        /**< whether set MAC address of target AP or not. Generally, station_config.bssid_set needs to be 0; and it needs to be 1 only when users need to check the MAC address of the AP.*/
    uint8 bssid[6];         /**< MAC address of target AP*/
};

bool wifi_station_set_config(struct station_config *config)
```
note <b>uint8 bssid[6];         /**< MAC address of target AP*/</b>
So my inquiry is: Can we do the same with Simba, or is that not implemented yet? Are there any plans to implement it? My C is terrible, but I can try to add this functionality myself.

eerimoq, 2017-04-14:
Ahh ok.. no, that is not implemented yet. Would be wonderful if you can implement it. I can support if needed.

eimiss, 2017-04-14:
Yes. I'd be happy to. Will invest some time this weekend.

eimiss, 2017-04-15:
Hey Erik,
I stumbled upon a curiosity
```C
//network_interface.h
int network_interface_stop(struct network_interface_t *netif_p);
```
That function is declared and documented, but unimplemented?

Loving this framework so far, I'm not a programmer by profession but I appreciate the conciseness and tidiness of Simba :+1:

eerimoq, 2017-04-15:
Hmm... right, I must have forgotten about that function =)

I'm glad you like it, I did my very best to keep the code simple and easy to understand. I think the network interface module is a bit of a mess and not finalized, but in general I'm pretty happy with the Simba code base.