eerimoq/simba #96: Unclear MQTT client usage

simonlbn, 2017-04-14:
Hi,

I'm looking a sending data using MQTT, so I have been looking at the Simba mqtt_client. Reading the code and the test I'm not sure exactly how to use it.

It looks like mqtt_client provides only the protocol, and then the (socket) transport is separate - is that correct? If so, do you know of any existing code to use the Simba MQTT client over TCP (or even better TLS)? While I'm sure it would be instructive to code, I'm even more happy to not code things :-).

Regards,
Simon

eerimoq, 2017-04-15:
You're right, the mqtt_client module only implemnents the protocol, the transport is any stream channel, often a TCP socket.

A testcase creates a TCP socket and initializes the mqtt client. See https://github.com/eerimoq/simba/blob/master/tst/inet/mqtt_client_network/main.c#L82-L94.

I've not tested the client over TLS. The [SSL/TLS wrapper socket](https://github.com/eerimoq/simba/blob/master/src/inet/ssl.h) implements the channel API, so it should work.

Erik

simonlbn, 2017-04-16:
OK, thanks. I will have a go at that and when I get a bit more experience with it try and create an example.

I sent a pull request to make the channel part a bit more clear for now.

closed 2017-04-16 by eerimoq