eerimoq/simba #98: Arduino SAM - i2c

pctj101, 2017-04-16:
Wondering if anyone started work on the Arduino Due i2c driver.

Started to write out basic #defines for the SAM.
https://github.com/pctj101/simba/commit/bd7ff576e996fea6d5bec4682d239b8b98748947

I see a mismatch in that the SAM MCU has a concept of i2c internal addresses and seems to "rather than manually send which address to read", will allow one to configure the internal address of the slave i2c chip ahead of time.  Then on read (perhaps even via Peripheral DMA), the i2c address and the subsequent internal address seem to be automatically sent (rather than manually "byte banged" write-then-read)

http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-11057-32-bit-Cortex-M3-Microcontroller-SAM3X-SAM3A_Datasheet.pdf
```
Figure 33-11. Master Write with One, Two or Three Bytes Internal Address and One Data Byte 
Figure 33-12. Master Read with One, Two or Three Bytes Internal Address and One Data Byte
33.8.7 Using the Peripheral DMA Controller (PDC)
```

For example, with "Figure 33-17. TWI Write Operation with Multiple Data Bytes with or without Internal Address" Data is typically written then ends of with a TWI_CR = STOP.

It's almost as if the SAM i2c peripheral wants to treat any read/write "conversation" as an "atomic" transaction.

However, the Simba-OS API is more aligned with managing this "conversation" byte by byte.
https://simba-os.readthedocs.io/en/latest/library-reference/drivers/i2c.html

(The AVR was more byte based: http://www.atmel.com/images/Atmel-8271-8-bit-AVR-Microcontroller-ATmega48A-48PA-88A-88PA-168A-168PA-328-328P_datasheet_Complete.pdf - 22.7.6 Combining Several TWI Modes)

Any thoughts as to how I should proceed?  It's almost as if this MCU wants me to write functions such as:

addr = i2c addr
iaddr = internal address of i2c device
i2c_read_atomic(0xaddr, 0xiaddr, &readbytes, len)
i2c_write_atomic(0xaddr, 0xiaddr, &writebytes, len)

Not sure if I'm duplicating effort so let me know.

pctj101, 2017-04-16:
Maybe make a different `i2c_driver_t` per i2c internal address?

eerimoq, 2017-04-16:
Hi,

I'm not quire sure I understand why the SAM internal address has to be part of the API. I think it's a SAM specific construct that is not part of the I2C standard, and that the SAM port of the driver can be implemented without it exported in the API. But I may very well be wrong... =)

I prefer to keep the Simba i2c API as it is today unless a change is required to make i2c work on SAM.

pctj101, 2017-04-16:
Just FYI I tried a quick run of i2c_soft on the Due and it's not working consistently. (Target chip locks up after 1 run)

So... yeah I'm trying to get i2c working on the SAM.

I think I'm going to start by adding some *_port_* functions first and seeing if I can even get it to work (while avoiding the main API points).

If I can get it to work, then... I'll see what you think. :)

eerimoq, 2017-04-16:
Great =)

pctj101, 2017-04-16:
Hello again!

I could really use a second pair of eyes.  I've been trying to debug this for a number of hours but can't seem to make any more headway.

In this file, I'm trying to thrd_resume_isr(), but it seems to lock up my system:
https://github.com/pctj101/simba/blob/sam_i2c_v1400/src/drivers/ports/sam/i2c_port.i#L206
https://github.com/pctj101/simba/blob/sam_i2c_v1400/src/drivers/ports/sam/i2c_port.i#L224

Of course, std_printf from ISR() seems to crash things as well, but I did so because it was the easiest way I could find to prove the ISR() was being run. (which it is! :) )
ISR Lockup @ https://github.com/pctj101/simba/blob/sam_i2c_v1400/src/drivers/ports/sam/i2c_port.i#L192

Would you be able to share some common reasons you found for thread_resume_isr() "crashing"?

Thanks!

```
#include "simba.h"
#include "kernel/types.h"

static THRD_STACK(shell_stack, 2048);
static struct shell_t shell;
static struct i2c_driver_t i2c;

void debug_sam() {
    uint8_t readbuf[16];
    i2c_init(&i2c, &i2c_device[1], I2C_BAUDRATE_100KBPS, -1);
    i2c_start(&i2c);
    std_printf(FSTR("read byte 0 begin\n"));
    i2c_read_txn(&i2c, 0x55, 0x00, 1, &readbuf, 16); /* Never returns */
    std_printf(FSTR("read byte 0 %d\n"), readbuf[0]);
}
int main()
{
    sys_start();
    shell_init(&shell, sys_get_stdin(), sys_get_stdout(), NULL, "simbashell", NULL, NULL);
    thrd_module_init();
    thrd_spawn(shell_main, &shell, 0, shell_stack, sizeof(shell_stack));

    /* Loop */
    while(1) {
      debug_sam();
      thrd_sleep_ms(2000);
    }
    return (0);
}

```

```
#define CONFIG_FS_CMD_THRD_LIST 1
#define CONFIG_SPI 0
#define CONFIG_MODULE_INIT_SPI 0
#define CONFIG_ADC 0
#define CONFIG_MODULE_INIT_ADC 0
#define CONFIG_MODULE_INIT_ANALOG_INPUT_PIN 0
#define CONFIG_MODULE_INIT_ANALOG_OUTPUT_PIN 0
#define CONFIG_I2C 1
#define CONFIG_MODULE_INIT_I2C 1
#define CONFIG_FS_CMD_I2C_READ 1
#define CONFIG_FS_CMD_I2C_WRITE 1
```

eerimoq, 2017-04-16:
Debugging from interrupt handlers are quite painful indeed, I usually add a `while(1);` in the interrupt handler to see that the program stops, and continue for there. Another approach is to use a hardware debugger, which most people don't have. Another alternative is to have a loop in a thread printing status flags and register values periodically. The status flags can be global variables written to from the interrupt handler.

Hard to say what is wrong, but maybe this helps:

1. `std_printf()` may not be used with the system lock taken (taken with `sys_lock()`): https://github.com/pctj101/simba/blob/sam_i2c_v1400/src/drivers/ports/sam/i2c_port.i#L486-L488

2. Maybe the register map is wrong.

3. I suggest that you remove as much code as possible from the interrupt handler to pinpoint the problem. Only keep the code related to an i2c write, for example.

pctj101, 2017-04-16:
I was taking a peek at how the Arduino folks do it... looks like they took a bunch of code from Atmel and plunked it in.

https://github.com/arduino/ArduinoCore-sam/tree/f5cbedd4e845b0fbb8165552f6ebb94363655234/system/libsam
https://github.com/arduino/ArduinoCore-sam/tree/f5cbedd4e845b0fbb8165552f6ebb94363655234/system/CMSIS

Sure seems like a reasonable way to go.. but I don't see it done much this way in this project.

Any thoughts?

eerimoq, 2017-04-16:
I prefer if you don't use a third party library for three reasons:

1. They are often not licensed under MIT.

2. They often bloated and does not fit in Simba's driver pattern.

3. Most drivers are not that hard to implement =)

pctj101, 2017-04-16:
Okay it looks like more than anything that the TWI peripheral in the SAM needs to get reset in some roundabout way before it starts acting "normal".

Will continue a different day.  Thanks!

pctj101, 2017-04-17:
I had a question about a possible race condition... just popped into my head.

Is it remotely possible that the following "race" occur?

1) I2C Interrupts are enabled
2) sys_lock
3) send first i2c byte 
  4) ISR called (earlier than expected) due to first byte successfully sent
  5) ISR calls thrd_resume_isr()
6) thrd_suspend_isr() is called (too late) which re-suspends the thread
7) now the thread is permanently locked

I don't yet completely understand which interrupts are enabled/blocked during sys_lock (couldn't be all since we still need the ISR of interest to fire right?)

Anyways, let me know if you happen to know off the top of your head.  

Thanks!

eerimoq, 2017-04-17:
All interrupts are disabled when `sys_lock()` has been called. At least that was the intention, and I'm pretty sure it works as expected.

This is how it works:

1. Call sys_lock() to disable interrupts.
2. Start the hardware job by writing to registers.
3. Call thrd_suspend_isr(). Interrupts are enabled with sys_unlock() once another thread has been scheduled, often the background idle thread.
4. Interrupt handler is called.
5. Once the job is complete, call thrd_resume_isr() from the interrupt handler.
6. The resumed thread is scheduled when prioritized.

eerimoq, 2017-04-17:
An alternative is to implement the driver without using interrupts, but instead poll the hardware in a while loop. When it works it can be converted to use interrupts.

pctj101, 2017-04-17:
Totally agree with polling first.  I'm trying it now and it doesn't work, so surely my code is at fault :)

eerimoq, 2017-04-17:
=)

pctj101, 2017-04-17:
I have no idea why, but I'm getting back all 0xFF on i2c reads.  It's the most crazy thing.

eerimoq, 2017-04-17:
Sometimes pull-up or pull-down resistors are needed to get the correct idle logical level on the wire.

pctj101, 2017-04-17:
I wanted to make that excuse, but the Wire.cpp (i2c) from the Arduino IDE works just fine with the exact same physical setup.  :(  I'm running out of things to blame.

pctj101, 2017-04-17:
So then I straight out removed simbaos and went straight with a .ino file with my code only... same crazy result.

So there's probably a 1 bit setup difference somewhere between my code and the Wire.cpp implementation.... wherever it is!

pctj101, 2017-04-18:
Did you know that printing a bunch of serial debug while trying to do time sensitive i2c reads all in the same single thread doesn't work?  I bet you already did.  :)

eerimoq, 2017-04-18:
Ehum that should work.. there is no reason it shouldn't if everything is done serially in the same thread =)

pctj101, 2017-04-18:
Hahaha... I think the "overly smart" SAM i2c peripheral is simply dumping bits over the side of the boat and clocking out bits while I print serial bytes.   It doesn't seem to wait for a read before reading the next byte, and there's only a 1 byte buffer... so... too late, overflow error. Removing the Serial.print (from arduino, not simbaos) resolves the issue.

Anyways, I have a little more testing to do and then I can get the code all merged back in the simba tree and send to you for review.

pctj101, 2017-04-18:
If it makes you laugh, with Serial.println in just the right places, I can read every other byte in the i2c stream.

pctj101, 2017-05-20:
@eerimoq Providing PR #116 for your review

eerimoq, 2017-07-03:
I2C driver for Arduino Due implemented in 143b47254b0aa03d8e7f832b5c43096b83134520.

closed 2017-07-03 by eerimoq