eerimoq/simba #101: Rust support

Kenzu, 2017-04-17:
Hey

What is the status of rust support in simba?
And how to use it?
As rust uses LLVM I assume it only work on non ESP platforms?

Thanks

Jesper

eerimoq, 2017-04-17:
Hi,

the Rust wrapper has not been used for a long time due to lack of intrest. The documentation is lacking and I can't remember how to install the prerequisites to make it compile. The Rust code itself probably works fine. You can give it a try, but some work will be needed to make it build.

Yeah, LLVM does not support ESP to my knowledge. I compiled for ARM (Arduino Due).

Kenzu, 2017-04-17:
Okay thanks...
We might take a look at it later.
Nice work with Simba and Pumbaa by the way :-)
Like it so far :-)

eerimoq, 2017-04-17:
Thanks! =)

Rust was quite intresting to use, but I lost interest when I failed to implement the `chan_list_poll()` function and had problems making good use of Rust without using dynamic memory allocations (which is discouraged in embedded application due to the memory constraints). =)

eerimoq, 2017-04-23:
@Kenzu I moved all Rust code to a [separate repository](https://github.com/eerimoq/rafiki) and wrote a [Getting Started guide](http://rafiki.readthedocs.io/en/latest/getting-started.html). The hello world example works on an Arduino Due. No other board is supported.

The Rust wrapper of Simba is barely started, but it shouldn't be too much work to implement wrappers of the kernel and sync packages. The most challenging part is to design an wrapper interface that is intuitive and easy to use. I failed last time, do you want to help out? =)

closed 2017-05-06 by eerimoq