eerimoq/simba #103: http_server_response_write() never returns if transmission is interrupted [ESP8266]

eimiss, 2017-04-18:
Hi!
Found a problem with the HTTP server. http_server_response_write() never returns if transmission is interrupted on client side - subsequent page requests are not being responded to. This is with ESP8266 - not sure whether other ports are affected. Steps to reproduce:
1) Flash the HTTP server example
2) Ctrl+C while curl is downloading page
```bash
eimis@e6430 ~/Projects/simba $ curl -i http://192.168.1.183
HTTP/1.1 200 OK
Content-Type: text/html
Content-Length: 9748

<!DOCTYPE html> <html> <head> <meta ^C
```
3) Subsequent curl calls never return or produce:
```bash
eimis@e6430 ~/Projects/simba $ curl -i http://192.168.1.183
curl: (56) Recv failure: Connection reset by peer
```
My guess is that chan_write() inside of http_server_response_write() never returns. I'd surely investigate this further but I'm not sure how to setup debugging on the ESP8266.
While http_server_response_write() is stuck ESP is still responsive via Simba shell.

eimiss, 2017-04-18:
Set up debugging, work to figure this out is in progress

eerimoq, 2017-04-19:
Just to let you know, ´std_printf()´ cannot be used in the callbacks that are executed in the LwIP thread. Use the ESP print function there, `printf()` if I remember correctly. Could be ´esp_printf()´ as well.

eimiss, 2017-04-23:
I'm stuck, Erik :grin: 
How do I make
```C
//http_server.c, line 108
log_object_print(NULL,
                     LOG_DEBUG,
                     OSTR("%s %s %s\r\n"), action_p, path_p, proto_p);
```
show up in the console output? Currently only LOG_INFO prints out. -DCONFIG_DEBUG=1 doesn't help

Also, printf produces no output at all regardless where it's called from
I'm unable to print from chan.c and event.c in any way. log_object_print() with LOG_DEBUG and std_printf() inside those files prevents boot - basically unable to troubleshoot this issue :/

It would be really great if I could find out where the http thread stops when page transfer is interrupted

eerimoq, 2017-04-23:
You can enable all log levels for the HTTP connection thread(s) with ``kernel/thrd/set_log_mask <http connection thread name> 0xff``.

``std_printf()`` cannot be used with the system lock taken, which is the case for the majority of the code in chan.c and event.c. Also, std_printf() does not work in _isr() functions called from the FreeRTOS LwIP thread. The LwIP callbacks in the socket module are called from the LwIP thread. Always use ``printf()`` in the LwIP callbacks.

eimiss, 2017-04-23:
Thank you, will try again. 
As I said though, **printf()** produces no output at all regardless where it's called from

eerimoq, 2017-04-23:
Comment out the lines in sys_port.i as described in my last port and it will give output.

eimiss, 2017-04-23:
Alright so thrd_suspend(NULL) is getting stuck in [socket.c line 180](https://github.com/eerimoq/simba/blob/master/src/inet/socket.c#L180)
Digging further :grin:

eimiss, 2017-04-23:
Seems like [```thrd_reschedule();```](https://github.com/eerimoq/simba/blob/master/src/kernel/thrd.c#L881) is not returning (thread state is ```THRD_STATE_CURRENT``` before ```thrd_suspend(NULL);``` is called). I can't print inside of [```thrd_suspend_isr()```](https://github.com/eerimoq/simba/blob/master/src/kernel/thrd.c#L852) so unable to verify.

If I change [socket.c line 180](https://github.com/eerimoq/simba/blob/master/src/inet/socket.c#L180) to ```return (thrd_suspend(&(struct time_t){.seconds = 3,}));``` then http server is able to serve further requests 3 seconds later after curl is interrupted. 1 second works too, but a microsecond cuts it out too early before whole response is sent. From my testing changing line 180 only has an effect when socket is closed - performance with uninterrupted transmissions in succession is unaffected.

I don't think that this is an acceptable solution. I either need a hint, or just have this patched on my fork and live with it. :white_flag:

eerimoq, 2017-04-23:
I had a quick look at the code, and it looks like the TCP close event from the LwIP stack is not handled correctly if received during an outgoing transmission. Not sure it is the case here, but you could try to resume the thread if the output socket state is ``STATE_SENDTO`` when ``pbuf_p`` is ``NULL`` in [``on_tcp_recv()``](https://github.com/eerimoq/simba/blob/master/src/inet/socket.c#L588-L596). Could also be an unhandled event in the [``err callback()``](https://github.com/eerimoq/simba/blob/master/src/inet/socket.c#L536-L548).

These are just best guesses after quickly looking at the code. The problem may very well be something else. I should write a testcase for your issue. Will try yo do it sooner rather than later as the socket module is very important for the ESP.