eerimoq/simba #107: PlatformIO Simba and Wemos Lolin32

f0m3, 2017-05-09:
When i try to build the examples on platformio i get a
`Environment arduino_mega        [SUCCESS]
Environment arduino_uno         [SUCCESS]
Environment arduino_nano        [SUCCESS]
Environment espressif_esp12e    [SUCCESS]
Environment lolin32             [ERROR]
========================== [ERROR] Took 75.87 seconds ==========================`

eerimoq, 2017-05-09:
What does platformio.ini look like?

Try this configuration:

```
[env:nano32]
platform = espressif32
framework = simba
board = nano32
```

The Wemos Lolin32 is not supported by Simba as of today. Would probably be easy to add, but I'm quite sure you can use the nano32 board instead. The MCU is the same on the two boards, and ESP32, right?

simonlbn, 2017-05-09:
For whatever it's worth, I ordered a WEMOS LOLIN32 when they were released and expect to add support for it to Simba when it arrives at some point (standard shipping from China to Europe... takes a while).

On a general note, it is another ESP32 based board with 4MB flash, so I would expect most things to work with one of other ESP32 board specification like eerimoq mentions.

f0m3, 2017-05-09:
I ordered one lolin32 from aliexpress 2 weeks ago and yesterday it arrived in Germany. I was impressed! :-) 
Building worked with 
```
[env:nano32]
platform = espressif32
framework = simba
board = nano32
```
I will try to upload tomorrow and report! Thanks for the help.

f0m3, 2017-05-10:
I changed &pin_led_dev to &pin_gpio05_dev in the simba/blink example and it worked! Thanks for the help!

closed 2017-05-10 by f0m3

eerimoq, 2017-05-10:
I updated the [porting documentation](http://simba-os.readthedocs.io/en/latest/developer-guide/porting.html) with more details on how to add a new board. Have a look there, and start be copying files from an existing ESP32 board, for example the Nano32.