eerimoq/simba #108: Example for mqtt

erenfro, 2017-05-11:
An example client would be very helpful. I noticed the framework has it's own mqtt_client feature built-in to the framework, and it would be very nice to see a good example of how to utilize it to connect to an mqtt server (plain and/or tls), subscribe to a topic, publish a topic, receive a subscription message, and also one of the kind of useful features of mqtt, setting and using the last will and testament feature of mqtt so that even in ungraceful disconnection, something still gets published to something that may be watching that situation.

eerimoq, 2017-05-11:
Hello,

[this testcase](https://github.com/eerimoq/simba/blob/master/tst/inet/mqtt_client_network/main.c) is the closest to an example as of today. Feel free to write an example and submit a pull request. That would be very helpful.

simonlbn, 2017-05-11:
For whatever it's worth, creating an MQTT example is still high on my personal TODO list as I ran into the same problem, but requires having actual spare time...

erenfro, 2017-05-11:
Heh. Yeah. I'd love to make an example, but for me, I need an example in order to understand how it's supposed to work. :)
Besides, I'm still new to C/C++, an example coming from me may not be that good. The test unit has just about everything except for receiving a message. For me to really be able to start using Simba, I needed two basic things to get started. One is to get connected to WiFi. That part I have, minimally, but have. The second is mqtt client support, which is where I'm trying to get focus on now.

Beyond that, one optional thing is logging to syslog, but I might be able to work something for that in by porting Arduino code over, eventually, but it's lower priority than mqtt simply because a majority of my work involves WiFi+MQTT, and the reason I'm looking into Simba as a whole is because I need TLSv1.2, not TLSv1.0 or TLSv1.1 that axTLS is limited to in current Arduino SDK, but TLSv1.2 which mbedTLS in Simba seems to support, which I'm happy about.

simonlbn, 2017-05-14:
Just FYI - I got a local MQTT broker up and running, and working with mqtt_client_network test.

I got example code in progress but as I'm learning MQTT and the Simba mqtt_client in the process it's not quite done yet...

eerimoq, 2017-05-14:
Another just FYI - the MQTT client module has not been used in any application other than the test suite (to my knowledge). If you face any problems when implementing the example it's probably due to bugs in the MQTT module.

simonlbn, 2017-05-14:
OK, thanks I will keep it in mind. That said, it does connect to the Mosquitto MQTT broker (which seems to be one of the widely used ones) and so far seems to just work with my simple tests.

eerimoq, 2017-05-14:
Then it works better than I could imagine, nice. :-)

closed 2017-05-30 by eerimoq