eerimoq/simba #109: MQTT Last Will and Testament

erenfro, 2017-05-11:
MQTT supports a feature called "Last Will and Testament". This is basically a client initiated, server-maintained negotiation of sorts. Essentially a topic/payload is provided in the case the mqtt client disconnects without negotiating the disconnection, such as in the case of a dropped connection, failure, network interruption, etc. 
After reviewing the mqtt_client.c code, I did not see any facilities for sending this last will feature. But everything else looks pretty feature complete more so than most embedded mqtt libraries I've seen. (very impressed!)

eerimoq, 2017-05-13:
Ohh okay, I didn't know about that feature. Would it be easy to add?

simonlbn, 2017-05-28:
I looked a bit into this, "unfortunately" the Last Will and Testament is sent to the MQTT broker at connection time, so the current Simba mqtt_client API has to be adjusted a bit to support that. It might make sense to do at the same time as a few other changes such as splitting init and start of driver.

eerimoq, 2017-05-30:
Feel free to make any changes, but of course, keep non-backwards compatible changes to a minimum to minimize the impact on any users.

PR #156 (2017-10-15): mqtt_client: Add support for connection options.
This allows setting client id, username, password, and Last Will and
Testament in MQTT connection options.

Setting this is done through a parameter to
mqtt_client_connect(). While this changes the API, it seems like the
best aproach as it allows the client not to have to keep any state for
the new options and therefor do not increase memory consumption other
than during connect.

The change is made so it's very easy for existing code to be updated,
as it simply need to change connect call from:

  mqtt_client_connect(client);

to:

  mqtt_clien [... cut]
closed 2017-10-15 by eerimoq