eerimoq/simba #110: MQTT authentication

erenfro, 2017-05-11:
The mqtt lacks the ability to authenticate to an mqtt server via username/password. For use in more sensitive situations, authentication may be needed to connect to an mqtt (or mqtts) server with authentication before it can proceed to communicate. Since MQTT is used often in IoT projects that may involve security sensitive situations, such as lock controls, door controls, etc, authentication is definitely strongly recommended.

eerimoq, 2017-05-13:
I've not used MQTT myself mor ethan in theory designing the MQTT module, and my approach was that the connection will be encrypted by SSL/TLS anyway, so a username/password would not be important to ensure data integrity. Anyway, I guess many brokers requires username/password, so it has to be implemented anyway. I'll not have time to implement it in the near future, sorry.

simonlbn, 2017-05-14:
I will need MQTT authentication for my current project, but no guarantees for when I will get to it.

simonlbn, 2017-07-03:
A small status update: I got the code to support authentication and last will and testament, and I'm not working on testing it and writing unit tests.

PR #156 (2017-10-15): mqtt_client: Add support for connection options.
This allows setting client id, username, password, and Last Will and
Testament in MQTT connection options.

Setting this is done through a parameter to
mqtt_client_connect(). While this changes the API, it seems like the
best aproach as it allows the client not to have to keep any state for
the new options and therefor do not increase memory consumption other
than during connect.

The change is made so it's very easy for existing code to be updated,
as it simply need to change connect call from:

  mqtt_client_connect(client);

to:

  mqtt_clien [... cut]
closed 2017-10-15 by eerimoq