eerimoq/simba #118: Websocket non-masked write option

skuhtic, 2017-05-21:
I tried websocket example with ESP32 (in pumbaa). When using python websocket client it works fine but when using browser and javascript then throws 'Masked frame from server' error. I've checked RFC/web and find out that server side communication to web browser should NOT be masked (and browser communication to websocket server MUST be).
I found that in simba src/inet/http_websocket_server.c file there is no write option without masking set. If I understood RFC correctly, implementing would be easy (clear INET_HTTP_WEBSOCKET_MASK in lines 207, 209, 214 and removing 4 MASK bytes from header in lines 226-230). 
I don't know should it be done with separate write method, with parameter to write method or websocket server init method and how to do that in pumbaa.
Really love the experience with pumbaa and appreciate all your effort.

eerimoq, 2017-05-22:
Thank you, glad you enjoy it =)

To minimize the effort for me I would gladly accept a pull request with the changes that implements the suggested changes. I've not read the websocket RFC is detail and I'm sure you are right in your observation.

eerimoq, 2017-07-04:
Fixed in 13ec21d48ff79d022d229a4adb36e07f0dc64905.

closed 2017-07-04 by eerimoq