eerimoq/simba #124: ESP32: update wifi lib

LupascuAndrei, 2017-06-27:
I see that simba is currently using an older version of https://github.com/espressif/esp32-wifi-lib
There are many fixes regarding wifi crashes and issues in the latest esp32 wifi lib 
https://github.com/espressif/esp-idf/releases

Is there a straight forward way to update only the wifi libs inside simba?
Thank you in advance.

eerimoq, 2017-06-27:
Hi,

ok, nice catch, I'll make an uplift to version 2.0 soon. Hopefully Espressif hasn't done too many non-backwards compatible changes. =)

LupascuAndrei, 2017-06-27:
That's great, thank you for the quick response ! 

Can it be an uplift to 2.1 since it was merged recently into master?
I managed to reproduce the issues on 2.0, but not on 2.1.

eerimoq, 2017-07-03:
I started the ESP-IDF uplift, but facing some problems with system startup on v2.1 and sockets v2.0. Will give an update here when I know more about the problems.

LupascuAndrei, 2017-07-03:
Please let me know if there's anything i can help with, i'll gladly do my best

closed 2018-02-11 by eerimoq