eerimoq/simba #131: App name with platformio

shacal, 2017-08-06:
Hmm, did not find any way to define APP_NAME variable with platformio.

PS: Love this RTOS, working with KNX, Hitachi LCD with I2C and some Oled displays.
Will provide pull's for drivers and clients at some point :)

PS: UART with parity and bit settings? Documents says nothing, but have not checked the source for them yet.

eerimoq, 2017-08-06:
This is what it looks like in the Sconscript that PlatformIO uses to build a Simba appliation: https://github.com/eerimoq/simba/blob/master/make/platformio.sconscript#L3963-L3964

Unfortunately I don't know how to set a variable in the Scons environment, I only use the GNU Make based build system, which I know much better.

You can of course modify the application name in the Sconscript, but it's a system wide change of course.

Lovely, more drivers are always welcome! =)

And, no, parity and bit settings are not part of the UART driver API, and not in the code either. Can be added of course. Any suggestions on what the API should be?

shacal, 2017-08-06:
Yeh, I think problem is Scons... maybe for backdoor is to check root of app and read APP.txt and name from there... of have an config that defines name. app_name in sysinfo is kinda "ugly" 🗡 

For parity, I currently have own function uart_start_config(*uart, int baud, int parity, int eq)

and then have parity defined with UART_8N1  etc like SERIAL_XXX in arduino framework.
Personally, I think uart_init should have config variable as uart_init(struct uart_driver_t *self_p, struct uart_device_t *dev_p, long baudrate, int config, void *rxbuf_p, size_t size)

shacal, 2017-08-06:
Hmm, and just after post I remembered that not all chips support configuration and always use 8N1.
Maybe have uart_configure(config) to have default (8N1) and port will give build error as not supported :)

eerimoq, 2017-08-06:
Yeah, and changing ``uart_init()`` would break backwards compatibility. An alternative is to add ``uart_set_*()`` functions for additional initialization of the UART object, which takes effect when ``uart_start()`` is called. For example, ``uart_set_parity(int parity)``.

shacal, 2017-08-06:
Created new issue for uart parity.
Let's keep this for app name.

rafacouto, 2017-08-06:
Just commited PR #133 in order to use APP_NAME variable from the environment. It is another way to define the application name different from checking a file from sources directory (as @shacal suggests) but less intrusive and easy to implement in _platformio.sconscript_.

shacal, 2017-08-07:
This was my first implementation also, but for platformio and multiple platforms (like windows) it is pain.
Bud id say good addition!

Remo Kallio

eerimoq, 2017-08-07:
In PlatformIO 3.4.1 the application name can be set in an ``extra_script``, without too much trouble.

**platformio.ini**

```
[env:arduino_uno]
platform = atmelavr
framework = simba
board = uno
extra_scripts = config.py
```

**config.py**

```
Import('env')

# Set the Simba application name.                                                                                                                                  
env.Replace(NAME='MyApplicationName')
```

shacal, 2017-08-07:
Good catch, not using 3.4.1 yet 🙈

Remo Kallio

eerimoq, 2017-08-07:
Anything stopping you? =)

shacal, 2017-08-07:
Nope, updating later tonight. 🙈

eerimoq, 2017-08-07:
=P

eerimoq, 2017-08-07:
Let me know if it works as expected. If it does, I'll add it to the documentation.

rafacouto, 2017-08-07:
@eerimoq Well, extra_script seems a better solution. Anyway, patch #133 is not intrusive and it also works in the case of _NAME_ is not defined (at least in command line environments; I don't know if it works in click-windowed OS's) :wink:

eerimoq, 2017-08-07:
Yeah, Windows sucks for programmers =)

BTW, ``platformio.sconscript`` is a generated file. Edit ``make/platformio/platforimio.py`` instead and I can accept the pull request.

shacal, 2017-08-07:
config.py works with win, mac and ubuntu 👍

eerimoq, 2017-08-07:
Wonderful! Thanks for testing! =P

eerimoq, 2017-08-07:
I added a section in the documentation: http://simba-os.readthedocs.io/en/latest/user-guide/build-system.html#platformio

closed 2017-08-07 by eerimoq