eerimoq/simba #132: Create uart_set_parity(uart_driver_t*, int parity)

shacal, 2017-08-06:
int parity can be one of these:
#define UART_PARITY_5N1 0x00
#define UART_PARITY_6N1 0x02
#define UART_PARITY_7N1 0x04
#define UART_PARITY_8N1 0x06
#define UART_PARITY_5N2 0x08
#define UART_PARITY_6N2 0x0A
#define UART_PARITY_7N2 0x0C
#define UART_PARITY_8N2 0x0E
#define UART_PARITY_5E1 0x20
#define UART_PARITY_6E1 0x22
#define UART_PARITY_7E1 0x24
#define UART_PARITY_8E1 0x26
#define UART_PARITY_5E2 0x28
#define UART_PARITY_6E2 0x2A
#define UART_PARITY_7E2 0x2C
#define UART_PARITY_8E2 0x2E
#define UART_PARITY_5O1 0x30
#define UART_PARITY_6O1 0x32
#define UART_PARITY_7O1 0x34
#define UART_PARITY_8O1 0x36
#define UART_PARITY_5O2 0x38
#define UART_PARITY_6O2 0x3A
#define UART_PARITY_7O2 0x3C
#define UART_PARITY_8O2 0x3E

shacal, 2017-08-06:
Must be called before uart_start and this keeps old code kicking 💃

eerimoq, 2017-08-06:
Parity is just on bit in an UART frame. I think we should call the function ``uart_set_format(int format)`` instead, and it takes one of the defines ``UART_FORMAT_XYZ`` as argument.

https://en.wikipedia.org/wiki/8-N-1
https://upload.wikimedia.org/wikipedia/commons/f/f2/UART_XBee_Data_format.jpg

shacal, 2017-08-06:
Thats why I was thinking uart_set_config(*d, int config)
Much simpler, than having parity, bits ets in own functions.

eerimoq, 2017-08-06:
Lol, yeah, I guess so =)

rafacouto, 2017-08-07:
@shacal Why are you using those int literals? I suggest to use bit encoding enums to mask with _bitwise and_ operations:

```
// UART modes. See https://en.wikipedia.org/wiki/Serial_port

#define UART_DATA_BITS_MASK 0b00000111
#define UART_DATA_BITS_5 0b00000001
#define UART_DATA_BITS_6 0b00000010
#define UART_DATA_BITS_7 0b00000011
#define UART_DATA_BITS_8 0b00000100
#define UART_DATA_BITS_9 0b00000101

#define UART_PARITY_MASK 0b00111000
#define UART_PARITY_N 0b00001000
#define UART_PARITY_O 0b00010000
#define UART_PARITY_E 0b00011000
#define UART_PARITY_M 0b00100000
#define UART_PARITY_S 0b00101000

#define UART_STOP_BITS_MASK 0b11000000
#define UART_STOP_BITS_0 0b01000000
#define UART_STOP_BITS_1 0b10000000
#define UART_STOP_BITS_2 0b11000000

#define UART_MODE_5N0 (UART_DATA_BITS_5 | UART_PARITY_N | UART_STOP_BITS_0)
#define UART_MODE_5N1 (UART_DATA_BITS_5 | UART_PARITY_N | UART_STOP_BITS_1)
#define UART_MODE_5N2 (UART_DATA_BITS_5 | UART_PARITY_N | UART_STOP_BITS_2)
#define UART_MODE_5O0 (UART_DATA_BITS_5 | UART_PARITY_O | UART_STOP_BITS_0)
#define UART_MODE_5O1 (UART_DATA_BITS_5 | UART_PARITY_O | UART_STOP_BITS_1)
#define UART_MODE_5O2 (UART_DATA_BITS_5 | UART_PARITY_O | UART_STOP_BITS_2)
#define UART_MODE_5E0 (UART_DATA_BITS_5 | UART_PARITY_E | UART_STOP_BITS_0)
#define UART_MODE_5E1 (UART_DATA_BITS_5 | UART_PARITY_E | UART_STOP_BITS_1)
#define UART_MODE_5E2 (UART_DATA_BITS_5 | UART_PARITY_E | UART_STOP_BITS_2)
#define UART_MODE_5M0 (UART_DATA_BITS_5 | UART_PARITY_M | UART_STOP_BITS_0)
#define UART_MODE_5M1 (UART_DATA_BITS_5 | UART_PARITY_M | UART_STOP_BITS_1)
#define UART_MODE_5M2 (UART_DATA_BITS_5 | UART_PARITY_M | UART_STOP_BITS_2)
#define UART_MODE_5S0 (UART_DATA_BITS_5 | UART_PARITY_S | UART_STOP_BITS_0)
#define UART_MODE_5S1 (UART_DATA_BITS_5 | UART_PARITY_S | UART_STOP_BITS_1)
#define UART_MODE_5S2 (UART_DATA_BITS_5 | UART_PARITY_S | UART_STOP_BITS_2)

#define UART_MODE_6N0 (UART_DATA_BITS_6 | UART_PARITY_N | UART_STOP_BITS_0)
#define UART_MODE_6N1 (UART_DATA_BITS_6 | UART_PARITY_N | UART_STOP_BITS_1)
#define UART_MODE_6N2 (UART_DATA_BITS_6 | UART_PARITY_N | UART_STOP_BITS_2)
#define UART_MODE_6O0 (UART_DATA_BITS_6 | UART_PARITY_O | UART_STOP_BITS_0)
#define UART_MODE_6O1 (UART_DATA_BITS_6 | UART_PARITY_O | UART_STOP_BITS_1)
#define UART_MODE_6O2 (UART_DATA_BITS_6 | UART_PARITY_O | UART_STOP_BITS_2)
#define UART_MODE_6E0 (UART_DATA_BITS_6 | UART_PARITY_E | UART_STOP_BITS_0)
#define UART_MODE_6E1 (UART_DATA_BITS_6 | UART_PARITY_E | UART_STOP_BITS_1)
#define UART_MODE_6E2 (UART_DATA_BITS_6 | UART_PARITY_E | UART_STOP_BITS_2)
#define UART_MODE_6M0 (UART_DATA_BITS_6 | UART_PARITY_M | UART_STOP_BITS_0)
#define UART_MODE_6M1 (UART_DATA_BITS_6 | UART_PARITY_M | UART_STOP_BITS_1)
#define UART_MODE_6M2 (UART_DATA_BITS_6 | UART_PARITY_M | UART_STOP_BITS_2)
#define UART_MODE_6S0 (UART_DATA_BITS_6 | UART_PARITY_S | UART_STOP_BITS_0)
#define UART_MODE_6S1 (UART_DATA_BITS_6 | UART_PARITY_S | UART_STOP_BITS_1)
#define UART_MODE_6S2 (UART_DATA_BITS_6 | UART_PARITY_S | UART_STOP_BITS_2)

#define UART_MODE_7N0 (UART_DATA_BITS_7 | UART_PARITY_N | UART_STOP_BITS_0)
#define UART_MODE_7N1 (UART_DATA_BITS_7 | UART_PARITY_N | UART_STOP_BITS_1)
#define UART_MODE_7N2 (UART_DATA_BITS_7 | UART_PARITY_N | UART_STOP_BITS_2)
#define UART_MODE_7O0 (UART_DATA_BITS_7 | UART_PARITY_O | UART_STOP_BITS_0)
#define UART_MODE_7O1 (UART_DATA_BITS_7 | UART_PARITY_O | UART_STOP_BITS_1)
#define UART_MODE_7O2 (UART_DATA_BITS_7 | UART_PARITY_O | UART_STOP_BITS_2)
#define UART_MODE_7E0 (UART_DATA_BITS_7 | UART_PARITY_E | UART_STOP_BITS_0)
#define UART_MODE_7E1 (UART_DATA_BITS_7 | UART_PARITY_E | UART_STOP_BITS_1)
#define UART_MODE_7E2 (UART_DATA_BITS_7 | UART_PARITY_E | UART_STOP_BITS_2)
#define UART_MODE_7M0 (UART_DATA_BITS_7 | UART_PARITY_M | UART_STOP_BITS_0)
#define UART_MODE_7M1 (UART_DATA_BITS_7 | UART_PARITY_M | UART_STOP_BITS_1)
#define UART_MODE_7M2 (UART_DATA_BITS_7 | UART_PARITY_M | UART_STOP_BITS_2)
#define UART_MODE_7S0 (UART_DATA_BITS_7 | UART_PARITY_S | UART_STOP_BITS_0)
#define UART_MODE_7S1 (UART_DATA_BITS_7 | UART_PARITY_S | UART_STOP_BITS_1)
[... 9 lines cut ...]
#define UART_MODE_8E1 (UART_DATA_BITS_8 | UART_PARITY_E | UART_STOP_BITS_1)
#define UART_MODE_8E2 (UART_DATA_BITS_8 | UART_PARITY_E | UART_STOP_BITS_2)
#define UART_MODE_8M0 (UART_DATA_BITS_8 | UART_PARITY_M | UART_STOP_BITS_0)
#define UART_MODE_8M1 (UART_DATA_BITS_8 | UART_PARITY_M | UART_STOP_BITS_1)
#define UART_MODE_8M2 (UART_DATA_BITS_8 | UART_PARITY_M | UART_STOP_BITS_2)
#define UART_MODE_8S0 (UART_DATA_BITS_8 | UART_PARITY_S | UART_STOP_BITS_0)
#define UART_MODE_8S1 (UART_DATA_BITS_8 | UART_PARITY_S | UART_STOP_BITS_1)
#define UART_MODE_8S2 (UART_DATA_BITS_8 | UART_PARITY_S | UART_STOP_BITS_2)

#define UART_MODE_9N0 (UART_DATA_BITS_9 | UART_PARITY_N | UART_STOP_BITS_0)
#define UART_MODE_9N1 (UART_DATA_BITS_9 | UART_PARITY_N | UART_STOP_BITS_1)
#define UART_MODE_9N2 (UART_DATA_BITS_9 | UART_PARITY_N | UART_STOP_BITS_2)
#define UART_MODE_9O0 (UART_DATA_BITS_9 | UART_PARITY_O | UART_STOP_BITS_0)
#define UART_MODE_9O1 (UART_DATA_BITS_9 | UART_PARITY_O | UART_STOP_BITS_1)
#define UART_MODE_9O2 (UART_DATA_BITS_9 | UART_PARITY_O | UART_STOP_BITS_2)
#define UART_MODE_9E0 (UART_DATA_BITS_9 | UART_PARITY_E | UART_STOP_BITS_0)
#define UART_MODE_9E1 (UART_DATA_BITS_9 | UART_PARITY_E | UART_STOP_BITS_1)
#define UART_MODE_9E2 (UART_DATA_BITS_9 | UART_PARITY_E | UART_STOP_BITS_2)
#define UART_MODE_9M0 (UART_DATA_BITS_9 | UART_PARITY_M | UART_STOP_BITS_0)
#define UART_MODE_9M1 (UART_DATA_BITS_9 | UART_PARITY_M | UART_STOP_BITS_1)
#define UART_MODE_9M2 (UART_DATA_BITS_9 | UART_PARITY_M | UART_STOP_BITS_2)
#define UART_MODE_9S0 (UART_DATA_BITS_9 | UART_PARITY_S | UART_STOP_BITS_0)
#define UART_MODE_9S1 (UART_DATA_BITS_9 | UART_PARITY_S | UART_STOP_BITS_1)
#define UART_MODE_9S2 (UART_DATA_BITS_9 | UART_PARITY_S | UART_STOP_BITS_2)
```

rafacouto, 2017-08-07:
WIP: I have implemented (not tested yet) [_uart_set_mode_ with for ESP32 port](https://github.com/eerimoq/simba/compare/master...rafacouto:feat-uart-modes?expand=1)

shacal, 2017-08-07:
This is much more production ready version.
Im curently in middle of KNX TP UART develop and was somehow shocked that this was missing from UART and I just did dirty hack for me to keep me going and to have proper version discused here.

I really like this aproach and waiting to get this to next release 👍🏻

Remo Kallio

eerimoq, 2017-08-07:
A few comments on the WIP:

1. ``uart_set_format()`` should only do ``self_p->format = format``, and ``uart_start()`` should modify hardawre register ``CONF0``.

2. The ``UART_FORMAT_*`` defines values should be port specific, allowing each port to define them to the actual hardware register value. No need for switch statements in the driver, just do ``regs_p->CONF0 = self_p->format`` in ``uart_start()`` (of course, extra handling of other non-format bits in ``CONF0`` may be required).

rafacouto, 2017-08-07:
1. `uart_start()` is [currently modifying `CONF0`](https://github.com/eerimoq/simba/blob/master/src/drivers/ports/esp32/uart_port.i#L191).
2. Right, so I will use hardware values directly maskable with uC registers 👍

eerimoq, 2017-08-07:
1. Yeah, I know, and that is the only function that should modify the register.

shacal, 2017-08-07:
This starts to feel right direction. 👍🏻

rafacouto, 2017-08-08:
Modifications [done](https://github.com/eerimoq/simba/compare/master...rafacouto:feat-uart-modes). I used a default initialization without modify the init function with new parameters and keeping legacy code compatible. This week I will found time to test in deep with ESP32 hardware 😄

shacal, 2017-08-08:
Make it as a pull? So can review?
Looks really good, need to test this out with atmega.

eerimoq, 2017-08-08:
What will the ``uart_read`` and ``uart_write()`` functions buffer contain when the number of bits in an UART frame does not contain 8 bits? I've no experiance with anything but 8 data bits, so I cannot really tell. Will bytes be zero padded, or simply still contain 8 data bits?

shacal, 2017-08-08:
Contain 8 bits, but 1 bit gets masked away.

shacal, 2017-08-08:
The 9bit version gets a bit messy :D

eerimoq, 2017-08-08:
Lol, ok, so what we have to do is to have a basic, easy to use, interface. If the user application has special requirements they have to add that functionality themselves, I think.

shacal, 2017-08-08:
That what arduino does and 7b and 9b is RARE. Even, odd and bits is most of the devices.

eerimoq, 2017-08-08:
Would be lovely if you guys can make changes to read/write functions as well (in case any changes are required) in the same pull request as the configuration changes.

shacal, 2017-08-08:
There is nothing to change in read/write.
Uart register is 8bit, we need to buffer it. Application needs to adapt to 7/9 bit if needed.

eerimoq, 2017-08-08:
Ok, so basically it the number of data bits are more than 8, all but 8 bits are simple overhead on the UART bus? For an UART with less than 8 data bits, the application will not use 8 - data bits MSB of the bytes in the read/write interface?

rafacouto, 2017-08-08:
Yes, I think there is no way to manage the minimal unit of data with microcontrollers (the byte). The rare thing is 9 bits and I have read that pics and avrs can use this format but not studied how. Anyway, it is independent from format and we should open another issue to that :wink:

eerimoq, 2017-08-08:
Sure, just have to make sure that the format configuration works as expected, as in discarding the MSB bits that does not fit in a byte.

shacal, 2017-08-08:
Most MCU skips 9th bit. It adds larger EOP and is mostly used for that. For data 7 and 8 is 99,5% of cases. So, this allows atlest for me to communicate 9bit version of modbus as it needs 9bits, but that bit is not data and is always LOW and is marked in protocol for extra EOP.

Even linux uart handles this 8bit always. ✌🏻

eerimoq, 2017-08-08:
Goodie, imaging I'm not here asking questions about an implementation that will work as expected =)

shacal, 2017-08-08:
For read, with 7 bits we might clear one bit to 7, but quick check looks that in atmel that bit is zero when in 7bit mode... so, lets save operations for other tasks 😝

rafacouto, 2017-08-08:
Oh no, travis says 

`/home/travis/build/eerimoq/simba/src/drivers/network/uart.c:97:11: error: ‘struct uart_driver_t’ has no member named ‘format’ self_p->format = format;`

The commit is not good. I am coding under platformio and I have desincronized with simba repository. Definitively, I have to mount the simba repo+toolchain 👼 

Please, can you take a look on [this error](https://travis-ci.org/eerimoq/simba/builds/262197644?utm_source=github_status&utm_medium=notification)?

shacal, 2017-08-08:
Platformio + git simba
Use app name config.py to define simba source...
Hmm, that wpuld be golden to work ☺️

eerimoq, 2017-08-08:
The design of the drivers are not perfect. Each port declares the uart_driver_t struct. You have to add int format here: https://github.com/eerimoq/simba/blob/master/src/drivers/ports/linux/uart_port.h#L40-L45

eerimoq, 2017-08-08:
@shacal I have ~/.platformio/packages/simba`` as a symbolic link to my simba git repo.

shacal, 2017-08-08:
Hmm, when we are in uart mode now.
Any good way to get isr from rx of uart?

eerimoq, 2017-08-08:
I could make the UART RX isr symbol weak so it can be overridden at link time, an alternative is to install a usr callback called on each interrrupt. So no, no good way in the current implementation.

shacal, 2017-08-08:
Weak is bad, as you cant call original one.
Callback with ret value (0 ok, 1 handled so no queue write).
As im working with strick timing protocol this would allow this out of box ✌🏻

eerimoq, 2017-08-08:
It will probably be under an ``ifdef`` since its not normal usecase, but I guess that is fine.

rafacouto, 2017-08-13:
State of art:
```
  Port   | Implemented |    Tested   |
---------|-------------|-------------|
avr      |     [ ]     |     [ ]     |
esp      |     [X]     |     [ ]     |
esp32    |     [X]     |     [X]     |
linux    |     [ ]     |     [ ]     |
nrf5     |     [ ]     |     [ ]     |
sam      |     [ ]     |     [ ]     |
samd     |     [ ]     |     [ ]     |
spc5     |     [ ]     |     [ ]     |
stm32f1  |     [ ]     |     [ ]     |
stm32f2  |     [ ]     |     [ ]     |
stm32f3  |     [ ]     |     [ ]     |
```

eerimoq, 2017-08-13:
@shacal Can you implement and test the AVR port once the pull request has been merged?

shacal, 2017-08-13:
Sure, AVR is in my table atm and can test couple variations from tiny to mega

rafacouto, 2017-08-20:
@shacal What about avr ports? After esp and stm, I could write the code quickly for avr's... Let me know if I can help in this task 😉

shacal, 2017-09-14:
Sorry for long delay...
Here is current "hack" that works for me : https://github.com/eerimoq/simba/pull/147

closed 2018-02-11 by eerimoq