eerimoq/simba #135: Master branch breaks linker with platformio

shacal, 2017-08-10:
Using latest master with platformio and seems that something is broken.
Release version works fine, but switching to master missing some references src/oam/console.c:61: undefined reference to `uart_start'

This happens in macOS, not tested others yet.

eerimoq, 2017-08-10:
``platformio.sconscript`` is a generated file that has not been generated recently. There is probably some inconsistency that a regeneration will solve.

``make platformio`` generates the file, but it requires the Simba build system to work, including toolchains.

shacal, 2017-08-10:
Oh, yeh, you moved drivers to subfolders.
scon can't find all files anymore :D

eerimoq, 2017-08-10:
Lol yeah that's it =)

I can generate a new one tonight if you havn't already generated one.

shacal, 2017-08-10:
Can't generate now, as mac make fails on me... cglang problem with stdio.h :(

eerimoq, 2017-08-10:
Pushed a fix.

shacal, 2017-08-10:
Cool, I try to test asap.

shacal, 2017-08-10:
A, btw any milestone for next release for platformio?
Build machines and auto deploy machines run PIO to build, test and publish... did not get any easy way to hack github version in :(

eerimoq, 2017-08-10:
Not sure, a few new drivers, NMEA sentence parser, probably some bug fixes, nothing major I think.

shacal, 2017-08-10:
Id like to include uart config with avr in that list :P

eerimoq, 2017-08-10:
Sure, if it is ready =)

eerimoq, 2017-08-10:
Ohh, there is a major change for AVR. The system tick uses a 16 bits hardware timer instead of 8 bits. So if someone is using that timer they'll be mad.

shacal, 2017-08-10:
I'm 🗡 
:P

eerimoq, 2017-08-10:
What do you mean??? =)

shacal, 2017-08-13:
Works with latest pull

closed 2017-08-13 by shacal