eerimoq/simba #136: Force Simba rebuild in PlatformIO

YegorBobrovski, 2017-08-10:
I am trying to fix issue discussed on forum ([link](https://forum.simbaos.org/showthread.php?tid=17)), but there is a problem.

Code generation for settings module is performed in Simba build task, what can be seen in output:
`Adding setting foo at address 0x0.`
`Adding setting bar at address 0x4.`

PlatformIO builds Simba only once until it registers some changes, for example, in platformio.ini file.
Is it available to register settings.ini file in PLatformIO to track changes for same behavior?

eerimoq, 2017-08-10:
PlatformIO uses a Python build framework called Scons. I have limited knowledge about that build system, but I'm sure it can somehow track changes in ``settings.ini``. Google for Scons and have a look at that documentation.

YegorBobrovski, 2017-08-18:
With SCons it can be done like this:
```
try:
    settings_ini = env.File(path, env['PROJECT_DIR'])
except Exception as e:
    sys.exit("Failed to load config file. " + str(e))

gen_command = env.Command(...)

if exists(str(settings_ini)):
    env.Depends(gen_command, settings_ini)
```

closed 2017-08-18 by YegorBobrovski