eerimoq/simba #144: STM32 conflicting BYTE_ORDER define

rafacouto, 2017-08-20:
Trying to compile blink example with stm32:

```
$ make -S BOARD=stm32vldiscovery

>>> app: blink, board: stm32vldiscovery <<<

make generate
make[1]: Entering directory '/home/caligari/src/simba/simba/examples/blink'
make[1]: Nothing to be done for 'generate'.
make[1]: Leaving directory '/home/caligari/src/simba/simba/examples/blink'
make build
make[1]: Entering directory '/home/caligari/src/simba/simba/examples/blink'
CC /home/caligari/src/simba/simba/3pp/compat/arch/sys_arch.c
mkdir -p build/stm32vldiscovery/obj/home/caligari/src/simba/simba/3pp/compat/arch
mkdir -p build/stm32vldiscovery/deps/home/caligari/src/simba/simba/3pp/compat/arch
mkdir -p build/stm32vldiscovery/gen
arm-none-eabi-gcc -I. -Ibuild/stm32vldiscovery/gen -I/home/caligari/src/simba/simba/src -I/home/caligari/src/simba/simba/src/boards/stm32vldiscovery -I/home/caligari/src/simba/simba/src/mcus/stm32f100rb -I/home/caligari/src/simba/simba/src -I/home/caligari/src/simba/simba/3pp/compat -I/home/caligari/src/simba/simba/src/drivers/ports/stm32f1 -I/home/caligari/src/simba/simba/3pp/lwip-1.4.1/src/include -I/home/caligari/src/simba/simba/3pp/lwip-1.4.1/src/include/ipv4 -I/home/caligari/src/simba/simba/3pp/mbedtls/include -I/home/caligari/src/simba/simba/src/kernel/ports/arm/gnu -I/home/caligari/src/simba/simba/src/oam/ports/stm32f1 -I/home/caligari/src/simba/simba/3pp/spiffs-0.3.5/src -I/home/caligari/src/simba/simba/3pp/atto -I/home/caligari/src/simba/simba/3pp/atto/curses -DCONFIG_MINIMAL_SYSTEM=1 -DCONFIG_PIN=1 -DCONFIG_LINUX_SOCKET_DEVICE=1 -DARCH_ARM -DFAMILY_STM32F1 -DMCU_STM32F100RB -DBOARD_STM32VLDISCOVERY -DVERSION=master -DMBEDTLS_USER_CONFIG_FILE="\"mbedtls/user_config.h\"" -DF_CPU=24000000UL  -O2 -Werror -mthumb -mcpu=cortex-m3 -ffunction-sections -fdata-sections -fpack-struct -c -Wall -funsigned-bitfields -std=gnu99 -Wno-error=unused-variable -Wno-error=deprecated-declarations -o build/stm32vldiscovery/obj/home/caligari/src/simba/simba/3pp/compat/arch/sys_arch.o /home/caligari/src/simba/simba/3pp/compat/arch/sys_arch.c
In file included from /home/caligari/src/simba/simba/3pp/lwip-1.4.1/src/include/lwip/arch.h:43:0,
                 from /home/caligari/src/simba/simba/3pp/lwip-1.4.1/src/include/lwip/debug.h:35,
                 from /home/caligari/src/simba/simba/3pp/lwip-1.4.1/src/include/lwip/opt.h:46,
                 from /home/caligari/src/simba/simba/3pp/lwip-1.4.1/src/include/lwip/init.h:35,
                 from /home/caligari/src/simba/simba/3pp/compat/arch/sys_arch.c:33:
/home/caligari/src/simba/simba/3pp/compat/arch/cc.h:52:0: error: "BYTE_ORDER" redefined [-Werror]
 #    define BYTE_ORDER LITTLE_ENDIAN
 ^
In file included from /usr/include/newlib/sys/types.h:67:0,
                 from /usr/include/newlib/stdio.h:48,
                 from /home/caligari/src/simba/simba/src/simba.h:61,
                 from /home/caligari/src/simba/simba/3pp/compat/arch/sys_arch.c:31:
/usr/include/newlib/machine/endian.h:20:0: note: this is the location of the previous definition
 #define BYTE_ORDER _BYTE_ORDER
 ^
cc1: all warnings being treated as errors
/home/caligari/src/simba/simba/make/app.mk:224: recipe for target 'build/stm32vldiscovery/obj/home/caligari/src/simba/simba/3pp/compat/arch/sys_arch.o' failed
make[1]: *** [build/stm32vldiscovery/obj/home/caligari/src/simba/simba/3pp/compat/arch/sys_arch.o] Error 1
make[1]: Leaving directory '/home/caligari/src/simba/simba/examples/blink'
/home/caligari/src/simba/simba/make/app.mk:110: recipe for target 'all' failed
make: *** [all] Error 2

```

It seems that **_BYTE_ORDER_ is defined twice** in:

- simba/3pp/compat/arch/cc.h
- /usr/include/newlib/sys/types.h

The second one is installed with the [system package libnewlib-dev](https://packages.debian.org/stretch/libnewlib-dev) (required by **libnewlib-arm-none-eabi**)

eerimoq, 2017-08-20:
We should create a Docker image with everything needed by Simba. It's the only feasible option to solve dependency issues.

rafacouto, 2017-08-20:
I agree with docker. New issue: #145

eerimoq, 2018-02-11:
Fixed.

closed 2018-02-11 by eerimoq