eerimoq/simba #145: Docker image with toolchain

rafacouto, 2017-08-20:
What docker image do you prefer to create the base? I suggest [debian:latest](https://hub.docker.com/_/debian/) but it is just a personal option to propose a starting point 😈

eerimoq, 2017-08-21:
I only have experience with Ubuntu images, and they work very well. Debian should work as well I guess, lets try =)

rafacouto, 2017-08-21:
Yep, Ubuntu is a Debian with desktop steroids. I will prepare the Dockerfile soon 👍

rafacouto, 2017-08-25:
A first automated-build version (0.1.0) is ready to pull from docker registry. It includes 
- Debian Stretch 9.1
- ESP32 SDK 1.22.0-59
- ESP8266 Open SDK 350c0e9 on Jun 7

It is possible to make some targets with `docker run -ti --rm caligari/simbabuilder bash -c 'cd simba ; source setup.sh ; make'`.

Next step is to make entry-point scripts to:
- specify the board and project directory to build the firmware.
- upload the firmware from the container (requires to bind devices and probably special docker permissions).

eerimoq, 2017-08-25:
Very nice, I'll give it a try in the weekend.

eerimoq, 2017-08-29:
Is the simba repository part of the image? Maybe preferable for simba users, but not for simba developers. Which group is the image targeting? =)

rafacouto, 2017-08-29:
Uhmmm, currently, it is simba users (my idea was to develop firmware with the simba framework). Anyway, let me think if a dual solution is possible just with one image. May be downloading / updating the simba repository at entry point. I see your suggestion, it should solve the problem of toolchain dependencies first...

eerimoq, 2018-02-11:
@rafacouto What's the status of this?

closed 2018-07-05 by eerimoq