eerimoq/simba #150: Succession: Low level driver vs driver

shacal, 2017-09-20:
Hi!

I'm seeing all drivers in same folder and that can clutter pretty fast.
I think that low level drivers (USART, SPI, I2C...etc) should be separate from high level "drivers" like "MQTT, DHT...etc".

Also, to have some kind to submodule support to inject more drivers would be nice. Like, having own driver, then git submodule that in and then have it included in pipeline.

What you guys think?

eerimoq, 2017-09-20:
Have a look in the drivers folder on the master branch, the drivers are split into a few categories.

MQTT is in the inet package, so that was already outside the drivers folder. DHT is now in the sensors folder. There are certainly opportunities to improve this further, but it's at least not all drivers in the same folder on master =)

Some kind of submodule concept is out of scope of the project. Instead, either create the new driver in your local simba repository, or in a separate repository. I'm sure it's easiest to do it in the simba repository since the build system gives drivers special treatment with `PORT_HAS_` defines. The configuration of supported drivers for each board could be improved for sure, but it's not planned since the current approach works for the use cases I've seen.

simonlbn, 2017-10-10:
One side comment - given how the current Simba driver structure works, as far as I recall it works perfectly fine to simply have a custom driver in your project directory and add it like any other .c file(s).

At least I'm fairly certain that was exactly how I did a good part of the development for the two drivers I have written (though I must admit it's a while ago so I could misremember).

eerimoq, 2017-10-11:
You are right, adding another .c file to the make variable `SRC` works perfectly fine.

closed 2018-02-03 by eerimoq