eerimoq/simba #151: ESP8266 i2c_read is returning -1.

rahuldeo2047, 2017-09-29:
Please follow: https://github.com/eerimoq/simba/issues/12

The board file updated accordingly by adding the following:
```
#define i2c_0_dev                                i2c_device[0]
#define i2c_1_dev                                i2c_device[1]

```

rahuldeo2047, 2017-09-29:
To test following modification:

```
ssize_t i2c_soft_read(struct i2c_soft_driver_t *self_p,
                      int address,
                      void *buf_p,
                      size_t size)
{
    size_t i;
    uint8_t *b_p;

    b_p = buf_p;

    /* Send the start condition. */
    if (start_cond(self_p) != 0) {
        return (-1);
    }

    /* Write the address iwth the direction bit set to 1. */
    if (write_byte(self_p, ((address << 1) | 0x1)) != 0) {
        stop_cond(self_p);
        return (-1);
    }

    // As per MPU6050 datasheet study : Exp
    #Warning i2c driver code modified: fn i2c_soft_read
    /* Send the start condition. */
    if (start_cond(self_p) != 0) {
        return (-1);
    }
          
    /* Read the data. */
    for (i = 0; i < size; i++) {
        if (read_byte(self_p, &b_p[i]) != 0) {
            stop_cond(self_p);
            return (-1);
        }
    }

    /* Send the stop condition. */
    if (stop_cond(self_p) != 0) {
        return (-1);
    }

    return (size);
}
```

rahuldeo2047, 2017-09-29:
Did not read :(

rahuldeo2047, 2017-09-29:
Reverted.
The stop_cond(...) is not working.

rahuldeo2047, 2017-09-29:
Checking if read ACK/NACK can be modified for.

```
static int read_byte(struct i2c_soft_driver_t *self_p,
                     uint8_t *byte_p)
{
    uint8_t bit;
    int i;

    bit = 0;
    *byte_p = 0;

    for (i = 0; i < 8; i++) {
        if (read_bit(self_p, &bit) != 0) {
            return (-1);
        }

        *byte_p = ((*byte_p << 1 ) | bit);
    }

    /* Acknowledge that the byte was successfully received. */
    if (write_bit(self_p, 0) != 0) { // ??????
        return (-1);
    }

    return (0);
}

```

rahuldeo2047, 2017-09-29:
Following seems to work for both single and burst mode data read:

```
/* Read the data. */
    for (i = 0; i < size; i++) {
        if (read_byte(self_p, &b_p[i], (i==(size-1))) != 0) {
            stop_cond(self_p);
            return (-1);
        }
    }
```

Func:

```
/**
 * Read a byte from I2C bus.
 */
static int read_byte(struct i2c_soft_driver_t *self_p,
                     uint8_t *byte_p, uint8_t nack )
{
    uint8_t bit;
    int i; 

    bit = 0;
    *byte_p = 0;

    for (i = 0; i < 8; i++) {
        if (read_bit(self_p, &bit) != 0) {
            return (-1);
        }

        *byte_p = ((*byte_p << 1 ) | bit);
    }

    /* Acknowledge that the byte was successfully received. */
    if (write_bit(self_p, nack) != 0) {
        return (-1);
    }

    return (0);
}

```

Please review the pull request for your convention. Coming in next.

rahuldeo2047, 2017-09-29:
When will it be released?

closed 2017-09-29 by rahuldeo2047

eerimoq, 2017-09-30:
Hello,

I'll gladly review any pull request once submitted.

/ Erik

rahuldeo2047, 2017-10-01:
Oh thank you for acknowledging. However, the fix which I was thinking to pull request, is already available in master. Which makes me question, when will it be released. :)
Secondly, do you have any plan to write/publish a general motion sensor driver? An example motion driver. General means using that driver other drivers can be written or extended in c way.

Let me know if this is not the correct forum.

eerimoq, 2017-10-01:
Last release was a few months ago and a lot of changes have been made since, most to the better I believe, so it's time for a new release soon. Maybe in a week or two, but I wont promise anything.

No I've no plans on a motion driver. What would it looks like? Would be lovely if you can create a pull request with a rough interface of the driver and we can discuss it from there.

rahuldeo2047, 2017-10-02:
Kool. It will require few other libraries like [science](http://simba-os.readthedocs.io/en/latest/library-reference/science/science.html), have some more functionalities. As per Issue #152. 
However, I'll attempt to put my best on it.

eerimoq, 2017-10-02:
Is that a question? =)

rahuldeo2047, 2017-10-03:
No :)