eerimoq/simba #152: Question: science module: vector, quaternion, matrix, statistics etc

rahuldeo2047, 2017-10-02:
Hi,
Can vector, quaternion, and matrix calculation routine/function be part of science library?
What are the plans for the [science](http://simba-os.readthedocs.io/en/latest/library-reference/science/science.html) library?

This issue relates to Issue #151. The mpu6050 sensor or similar might need these libraries.

Secondly; can this library have routine/function statistics, filters, DSP functions etc further?

eerimoq, 2017-10-02:
Sounds like a good match indeed. The plan is to add science related functionality to the package as the Simba project progresses.

Statistics, filters, etc. may very well be part of the science package. If needed, the science package can be split into smaller packages in the future.

rahuldeo2047, 2017-10-02:
@eerimoq: Thank you for your response. 

Can already available libraries be wrapped here?
e.g. wrapper for neural nets "[fann](http://leenissen.dk/fann/wp/)" etc.
As I am assuming so. Please correct me if otherwise.

Closing the question.
However this issues can be used to keep a wishlist, I guess. If @eerimoq agrees to it. :)

closed 2017-10-02 by rahuldeo2047

eerimoq, 2017-10-02:
The downside of making lots of libraries part of Simba is that the compile time will increase. On the other hand it's nice to make the libraries easily available to the user without too much build system fiddling. So far including more packages and modules has not been a problem.

rahuldeo2047, 2017-10-02:
ok