eerimoq/simba #155: Clarification needed: ESP: how to get elapsed time? Any otherway?

rahuldeo2047, 2017-10-07:
I checked the library code where it returns 0 for sys_uptime() and time_get(...) for ESP.

Need following steps to be implemented:
1. configuration
2. second
3. nanosecond
4. overflow handling
5. overhead calibration

---

Point 1. configuration: First need to go through rest of the points
Point 2. second: Existing is not working as expected, to confirm
Point 3. nanosecond: The estimated code is as below for nanoseconds part:

```
// requirement is nanoseconds
static int sys_port_get_time_into_tick()
{
    // get clock count
    // http://sub.nanona.fi/esp8266/timing-and-ticks.html
    uint32_t ccount;
    __asm__ __volatile__("esync; rsr %0,ccount":"=a" (ccount));

    // get nanos
    // from https://github.com/adafruit/ESP8266-Arduino/blob/50122289a3ecb7cddcba34db9ec9ec8b4d7e7442/hardware/esp8266com/esp8266/cores/esp8266/Esp.h#L129
    // system_get_cpu_freq() returns 80 or 160; unit MHz
    ccount /= system_get_cpu_freq(); // (freq_MHz*1000000) Hz = 1/second
    
    return ccount;
}

```
4. overflow handling: To speculate
5. overhead calibration: To speculate

Have these been implemented for ESP port?

eerimoq, 2017-10-07:
No there is no one actively working on ESP8266 timing precision.

In general, if a particular feature is missing on the master branch its not yet implemented.

rahuldeo2047, 2017-10-07:
OK

closed 2018-02-11 by eerimoq

rahuldeo2047, 2018-05-10:
Note: http://www.instructables.com/id/625-Nanosecond-Resolution-Timer-for-Any-Microcontr/