eerimoq/simba #160: Shell broken on at least ESP8266/ESP32 2017-07

simonlbn, 2017-10-19:
Hey,

The shell seems to have broken around July 17th and is still not working at head.

I tested using the shell example, and simply pressing enter once it started. When things worked I got the shell "$", and I didn't when it failed. I tested on a Wemos D1 Mini, and an ESP32 HUZZAH32 (using BOARD=esp32_devkitc). Those were the only boards I had handy, so I don't know if it effects non-ESP boards. 

I can't find the exact commit as there were some build failures around the time, but my binary search of errors found:

* 599ca05ad9c4010b896bb3e882de88d7aed85bc2 fail Mon Jul 17 18:57:39 2017 +0200
* 1c246833f6549b5dde982ef67eda0264d1017f85 fail Mon Jul 17 18:56:16 2017 +0200
* 1c3677be9a848abff840e02f8181a9b0fc3cab8c build fail
* 473a10c76b93138aa8f1e8be82462751a4f8e82e build fail Mon Jul 17 18:49:45 2017 +0200
* 52fc5ead85728274c5df5b2356b69d9e140e104e build fail Mon Jul 17 14:20:55 2017 +0200
* b789715e140cef6663ff4db0b6eea99a17f39188 build fail Mon Jul 17 14:14:43 2017 +0200
* 3b3524166ddce7b3740a7da9ed61af71d4512e4d ok Sun Jul 16 11:45:12 2017 +0200
* e233645935b4abfa36ade6a9f4ffa9b2bf1faf10 ok Sun Jul 16 11:30:39 2017 +0200
* 3b3524166ddce7b3740a7da9ed61af71d4512e4d ok Sun Jul 16 11:45:12 2017 +0200
* cf58308199548c9c4c8dc9b42d183a713e4aa34b ok Fri Jul 14 10:56:04 2017 +0200

I haven't yet looked more into the exact changes in the period, but there were changes which mentions UART changes so probably related?

eerimoq, 2017-10-20:
Hello,

The problem was introduced in [1c246833f6549b5dde982ef67eda0264d1017f85](Use UART base channel for both TX and RX to save memory.). Previously there was two channels in a UART driver, one for TX and one for RX. In this commit the two channels were merged into one to reduce memory usage. The problem is that received data is written to the channel using `chan_write_isr()`, which will call the registered tx-callback `uart_port_write_cb_isr()`. That is, received data is immediately transmitted on the UART instead of written to the shell. I fixed this in 9747152419e3059b22b599e826892799be5789d7 by replacing `chan_write_isr()` with `queue_write_isr()`.

simonlbn, 2017-10-20:
With that fix, it works again.

Thanks for the quick fix and the explanation.

closed 2017-10-20 by simonlbn