eerimoq/simba #161: Can't compile LCD Example using Platform IO, Blink and

ilovetogetspamed, 2017-10-20:
# Error 
```
...
Compiling .pioenvs/megaatmega2560/src/main.o

src/main.c: In function 'main':
src/main.c:36:29: error: storage size of 'lcd' isn't known
struct hd44780_driver_t lcd;
...
```
# Source
```
#define PORT_HAS_HD44780
#include "simba.h"

int main()
{
    struct hd44780_driver_t lcd;

    sys_start();

    hd44780_init(&lcd,
                 &pin_d8_dev,   // RS
                 &pin_d9_dev,   // ENABLE
                 &pin_d5_dev,   // DATA 4
                 &pin_d6_dev,   // DATA 5
                 &pin_d7_dev,   // DATA 6
                 &pin_d8_dev,   // DATA 7
                 2,             // ROWS
                 16);           // COLUMNS
    hd44780_start(&lcd);

    hd44780_display(&lcd, "Welcome!\r\n");
    hd44780_cursor_show(&lcd);

    while (1) {
        hd44780_put(&lcd, chan_getc(sys_get_stdin()));
    }

My LCD is wired a little differently (pin assignments are different) than your example but it should still compile I would think.

ilovetogetspamed, 2017-10-20:
Opening on PlatformIO Github site too.

eerimoq, 2017-10-21:
The HD44780 driver is not yet part of any release. You probably read the documentation for the master branch, and not for the release 15.0.3 which is found on PlatformIO.

The plan is to create a new release soon.

closed 2017-10-21 by eerimoq