eerimoq/simba #163: Detecting TCP connection closed

simonlbn, 2017-10-22:
Hi,

How do I determine if a TCP socket is still open?

For MQTT I would like to detect when the underlying TCP connection to the MQTT server/ broker fails, but it's not clear to me how to determine if the TCP connection is still open or not.

I spent some time reading the socket code, but nothing obvious came up yet.

With a pointer of how to find out about this I will be happy to document this (possibly make a helper function to return?).

Simon

eerimoq, 2017-10-22:
Hi,

the idea is that the read-function should return 0 or a negative error code if the peer closes the socket. It was quite some time since the socket module was implemented, and I don't remember if there are part of the functionality missing, or if everything is in place.

simonlbn, 2017-10-23:
OK, thanks. I will look more into it and try to understand how it actually works.

closed 2017-10-23 by simonlbn